meshtastic/firmware #758: Report Elevation while in range test,

dave032285, 2021-03-24:
Reporting elevation during the range test would be sweet. I guess it would be practical use for someone setting up on a mountain or something, personally im mounting meshtastic devices to my drones for impractical range testing. Maybe not impractical. But not real world testing. Unless it was for an emergency of some sort. Youd have an idea of how high you need to be to reach specific areas,

Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. TBEAM]
 - Software Version [e.g. 0.7.8]

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 0.7.2]

**Additional context**
Add any other context about the problem here.

geeksville, 2021-03-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-plugin-rangetestplugin/2591/66

dave032285, 2021-03-24:
Its actually a feature request. Did i post in the worng place?

geeksville, 2021-03-25:
@dave032285 no worries - that message wasn't really from me - it was just a side effect of your forum post linking here ;-)

mc-hamster, 2021-03-26:
This is done.

closed 2021-03-26 by mc-hamster

skycpdx, 2021-09-23:
Elevation and SNR fields (or headers) are backwards in the CSV

mc-hamster, 2021-09-24:
> Elevation and SNR fields (or headers) are backwards in the CSV

Thank you. I opened a bug for this.