meshtastic/firmware #760: Install/update scripts won't work with Python 3 in some distros

IZ1IVA, 2021-03-24:
**Describe the bug**
Install and update scripts expect a Python environment, but are unable to locate Python 3 in some distros such as Raspberry Pi OS o Debian.
Symlinks to /usr/bin/python are discouraged, however certain distros such as Ubuntu provide an optional python-is-python3 package which kinda does just that and preserves it across updates.

**To Reproduce**
Steps to reproduce the behavior:
1. Try executing the update or install scripts in Debian or Raspberry Pi OS
2. Observe the script failing:
`Trying to flash update firmware-tlora-v2-1-1.6-1.2.13.bin.`
`./device-update.sh: 42: ./device-update.sh: python: not found`
`Erasing the otadata partition, which will turn off flash flippy-flop and force the first image to be used`
`./device-update.sh: 44: ./device-update.sh: python: not found`
or
`Trying to flash firmware-tlora-v2-1-1.6-1.2.13.bin, but first erasing and writing system information`
`./device-install.sh: 44: ./device-install.sh: python: not found`

**Potential solutions**
1. Mandate Python 3 and modify the scripts accordingly
2. Make the scripts specifically look for the Python 3 executable whenever an older version is not found

geeksville, 2021-03-25:
@IZ1IVA  ooh good find!  Option 1 from your list seems fine to me.  Would you be willing to send in a PR?

IZ1IVA, 2021-03-25:
> @IZ1IVA ooh good find! Option 1 from your list seems fine to me. Would you be willing to send in a PR?

I just did it, though I'm definitely not a script guru! Props to @IZ1KGA for testing on Ubuntu.

geeksville, 2021-03-26:
fixed in #763

closed 2021-03-26 by geeksville