meshtastic/firmware #775: Option to connect via bluetooth if the smartphone has changed

Xanuman, 2021-03-30:
The situation is as follows.
I made a GPS tracker for sheep on the basis of the T-beam board and the Meshastic.
I sealed the board in a sealed enclosure. Only a cable with contacts for charging the battery is brought out. Those. there is no screen. There is no usb either.

If I change my smartphone or if the software fails, I will have to destroy the hermetic shell of the tracker to get a new Bluetooth connection code.

Is it possible somehow to create a permanent bluetooth connection code?
Or otherwise solve this problem

mc-hamster, 2021-04-03:
@geeksville With the new admin channel, would it be possible to send a secured message over the mesh network to instruct a given node to set the bluetooth connection code to defaultBLEPin until the device is next power cycled?

geeksville, 2021-04-03:
Oh good idea.  That would be pretty easy.

(Sent from a phone - please ignore typos)

garthvh, 2022-07-03:
Closing in favor of #1548

closed 2022-07-03 by garthvh