meshtastic/firmware #779: (Request) Add examples for HardwarePlugin

phil71994, 2021-04-05:
This is not a bug. I just discovered that meshtastic with its plugin support can actually handle GPIOs, but with the existing HardwarePlugin, i cant get my head around how i could handle a certain message to switch on or off a GPIO pin. Help would be greatly appreciated!

geeksville, 2021-04-05:
yeah good point - we really need to make some decent docs on the gpio stuff.  It was originally a proof of concept for the plugin API .

geeksville, 2021-04-05:
(Though btw - if you aren't already on the forum: meshtastic.discourse.group is also useful)

sachaw, 2021-04-05:
Just want to note, many users get confused between the concepts of the on device api and plugin system and the consumption of these with the various clients/libraries that we have. This is something we hope to clear up on the docs/getting started section

geeksville, 2021-04-06:
btw - In making this doc just now GPIOs were super broken in 1.2.  All fixed I think now ;-).  Will need device code 1.2.23 and python code 1.2.23

commit cbfd80f8 (2021-04-06): fix #779.  Add basic GPIO documentation
closed 2021-04-06 by geeksville