meshtastic/firmware #781: Pass power source in --info response

AvronW, 2021-04-06:
Referring to: https://meshtastic.discourse.group/t/how-can-i-determine-if-a-remote-node-is-on-usb-power-or-battery-using-the-cli/3080

It'd be great to be able to use a CLI call like _**–info**_ to see all mesh nodes and be able to determine from that which nodes are on USB power vs internal battery, not just the power level. This would allow us to identify if a solar charged battery has dropped to a level low enough that the USB power is no longer sufficient to power the device and the device has switched to its internal battery.

garthvh, 2022-01-22:
100 is fully charged on battery and 0 is powered

garthvh, 2022-01-24:
Closing this it is a display issue in the python api if this is still a desired feature.

closed 2022-01-24 by garthvh