meshtastic/firmware #785: Sound on message receipt

karl-baker, 2021-04-11:
Is it possible for the T-Beam device to play a sound or make some sort of noise when a message has been received?   If possible, could this be enhanced into a future firmware release?

MrSniffer, 2021-04-12:
I plan on using meshtastic for relaying emergency wildfire information in a rural community. Since power, phones, wifi, cell service all have been known to fail days into a wildfire, I am targeting battery powered standalone operations as a worst case scenario. So I will be attaching a sounder to T-Beams (or other) and have a need for the packet's to have some form of priority level (at least one emergency bit) which will sound the alarm at the high level. Mobile devices or PC will be connected during normal operations but can't be relied upon during the height of a crisis.

Chiumanfu, 2021-04-15:
https://github.com/meshtastic/Meshtastic-device/blob/master/docs/software/plugins/ExternalNotificationPlugin.md

closed 2021-10-23 by sachaw