meshtastic/firmware #787: Serial size the http data over to mynodeinfo

mc-hamster, 2021-04-12:
This was suggested by @sachaw during the Meshtastic Dev Conference.

garthvh, 2022-09-11:
@sachaw I still don't totally understand this one

sachaw, 2022-09-11:
Yeah, it's not named very clearly.
What it's inferring is that http exclusive features. Range test CSV,  file system browsing etc should be represented with protobufs so they can be accessed over any connection method.

garthvh, 2022-09-11:
https://github.com/meshtastic/Meshtastic-device/issues/1389#issue-1207732505

caveman99, 2022-09-12:
I vote for a generic file send/receive over protobufs so we can provision devices by putting proto files into the flash filesystem. One use would be the oem.proto used for a secondary bootlogo and text.

garthvh, 2023-02-11:
Is this complete @caveman99?

caveman99, 2023-02-11:
The implementation is done, but there's room for improvement. I think we can close this.

closed 2023-02-11 by garthvh