meshtastic/firmware #794: Remote "--dest" commands time out

Chiumanfu, 2021-04-17:
Sending remote commands to the admin channel using the --dest command times out.

See thread for more info and screenshots
https://meshtastic.discourse.group/t/admin-channel-timeouts-and-unresponsiveness/3146

Device 1.2.28
Python 1.2.30

geeksville, 2021-04-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/admin-channel-timeouts-and-unresponsiveness/3146/5

ghettodev, 2021-04-17:
Error:

```bash
➜  firmware-1.2.28 meshtastic --port=/dev/ttyUSB0 --dest \!050d5b74 --info --reboot                                                                                                                                                                                                   130 
Connected to radio
INFO:root:Requesting preferences from remote node (this could take a while)
ERROR:root:Error while handling message from radio 'admin'
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 844, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 531, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 669, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/node.py", line 291, in onResponse
    self.radioConfig = p["decoded"]["admin"]["raw"].get_radio_response
KeyError: 'admin'
Aborting due to: Timed out waiting for node config
```

meshtastic info:

```bash
➜  firmware-1.2.28 meshtastic --port=/dev/ttyUSB0 --info
Connected to radio

Owner: a7b0 (a7b)

My info: { "myNodeNum": 2442635184, "numBands": 13, "firmwareVersion": "1.2.28", "rebootCount": 32, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }

Nodes in mesh:
  {'num': 2442635184, 'user': {'id': '!9197a7b0', 'longName': 'a7b0', 'shortName': 'a7b', 'macaddr': 'UAKRl6ew', 'hwModel': 'TLORA_V2'}, 'position': {'latitudeI': 2, 'longitudeI': -2, 'altitude': 36, 'latitude': 2.2, 'longitude': -2.2}, 'lastHeard': 1618636346}
  {'num': 84761460, 'user': {'id': '!050d5b74', 'longName': '5b74', 'shortName': '5b7', 'macaddr': 'PGEFDVt0', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 2, 'longitudeI': -2, 'altitude': -1, 'batteryLevel': 100, 'time': 1618636197, 'latitude': 2, 'longitude': -2}, 'lastHeard': 1618636171, 'snr': 6.75}
  {'num': 84761736, 'user': {'id': '!050d5c88', 'longName': '5c88', 'shortName': '5c8', 'macaddr': 'PGEFDVyI', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 2, 'longitudeI': -2, 'altitude': 61, 'batteryLevel': 100, 'time': 1618636200, 'latitude': 2.2, 'longitude': -2.2}, 'lastHeard': 1618636174, 'snr': 7.0}
  {'num': 2442725896, 'user': {'id': '!91990a08', 'longName': '0a08', 'shortName': '008', 'macaddr': 'UAKRmQoI', 'hwModel': 'TLORA_V2', 'raw': id: "!91990a08" long_name: "0a08" short_name: "008" macaddr: "P\002\221\231\n\010" hw_model: TLORA_V2 }, 'position': {'latitudeI': 2, 'longitudeI': -2, 'altitude': -2, 'time': 1618636082, 'latitude': 2.2, 'longitude': -2.2}, 'lastHeard': 1618636319, 'snr': 6.0}

Preferences: { "positionBroadcastSecs": 360, "waitBluetoothSecs": 28800, "screenOnSecs": 360, "lsSecs": 300, "wifiSsid": "2", "wifiPassword": "2", "region": "US", "extNotificationPluginEnabled": true, "extNotificationPluginOutput": 14, "extNotificationPluginActive": true, "extNotificationPluginAlertBell": true, "storeForwardPluginRecords": 100, "storeForwardPluginEnabled": true }
```

`meshtastic --version: 1.2.30`
`python --version: Python 3.8.5`

IZ1IVA, 2021-04-20:
In my tests, admin mode only works when all nodes are awake before sending out the command and are still awake when the nearest node transmits or relays the command.

geeksville, 2021-08-02:
I think I fixed this some time ago and everyone was again happy.  Please reopen if this is still happening for you.

closed 2021-08-02 by geeksville

IZ1IVA, 2021-08-07:
> I think I fixed this some time ago and everyone was again happy. Please reopen if this is still happening for you.

Just checked: it's still happening :-(
So far I've only tested with a TTGO LoRa32 hooked to a Windows 10 PC trying to send commands to a T-Beam well within range (+2 dB SNR). Both boards are on 1.2.43a.

Cheers