meshtastic/firmware #797: SH1106 fails to build

kylegordon, 2021-04-22:
From, I believe, 1.2.20 onwards, it appears that building with USE_SH1106 causes it to fail with the attached messages. This seems to be the case all the way up to master branch.

Compiling .pio/build/tbeam/src/esp32/main-esp32.cpp.o
Compiling .pio/build/tbeam/src/gps/Air530GPS.cpp.o
Compiling .pio/build/tbeam/src/gps/GPS.cpp.o
Compiling .pio/build/tbeam/src/gps/NMEAGPS.cpp.o
Compiling .pio/build/tbeam/src/gps/RTC.cpp.o
Compiling .pio/build/tbeam/src/gps/UBloxGPS.cpp.o
In file included from src/esp32/BluetoothSoftwareUpdate.cpp:4:0:
src/esp32/../graphics/Screen.h:288:5: error: 'SH1106Wire' does not name a type
     SH1106Wire dispdev;
     ^
*** [.pio/build/tbeam/src/esp32/BluetoothSoftwareUpdate.cpp.o] Error 1
========================================================================================================= [FAILED] Took 4.54 seconds =========================================================================================================

Environment    Status    Duration
-------------  --------  ------------
tbeam          FAILED    00:00:04.544

patrick-rt, 2021-04-26:
same problem here

patrick-rt, 2021-04-26:
[image]
Looks like you can "fix" this by commenting out these lines in the Screen.h file. Worked for me

kylegordon, 2021-05-17:
Excellent, yes, commenting out those lines makes it work for the SH1106. That'll get me by for now.

I won't close this issue though... the problem still exists :-)

closed 2021-08-02 by geeksville