meshtastic/firmware #807: Errno 104 - TCP problems with meshtastic 1.2.33 and 1.2.30 firmware

pwilen, 2021-05-24:
when executing "meshtastic --host 192.168.68.xxx --info" I get:
ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer

I have tested with a T-beam and a Heltec. The units  are connected to my WiFi-router and I can access their (limited) web interface 192.168.68.xxx/static/index.html

Tested on raspbian and OS X

closed 2021-05-24 by geeksville

geeksville, 2021-05-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-release-of-python-api-1-2-35-and-geeksvilles-current-queue/3398/1

kallisti5, 2025-04-01:
Seeing the same issue lingering on meshtastic v2.5.4..

```
$ /usr/bin/python3 /home/kallisti5/meshweather.py
connecting to Meshtastic
ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer
Connected to Meshtastic device, my node ID: !daXXXXXX, number: 366403XXXX
Sending weather data to Meshtastic
XXX Weather 4/1/2025
  Today: ? 84F - 0 to 15 mph/SSE
Tonight: ? 66F - 10 to 15 mph/S
Traceback (most recent call last):
  File "/home/kallisti5/meshweather.py", line 60, in <module>
    interface.sendText(message)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/home/kallisti5/.local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 436, in sendText
    return self.sendData(
           ~~~~~~~~~~~~~^
        text.encode("utf-8"),
        ^^^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
        channelIndex=channelIndex,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/kallisti5/.local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 551, in sendData
    p = self._sendPacket(meshPacket, destinationId, wantAck=wantAck, hopLimit=hopLimit, pkiEncrypted=pkiEncrypted, publicKey=publicKey)
  File "/home/kallisti5/.local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 929, in _sendPacket
    self._waitConnected()
    ~~~~~~~~~~~~~~~~~~~^^
  File "/home/kallisti5/.local/lib/python3.13/site-packages/meshtastic/mesh_interface.py", line 1073, in _waitConnected
    raise MeshInterface.MeshInterfaceError(
        "Timed out waiting for connection completion"
    )
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for connection completion
```

Device is Heltec WSL v3

commit be410a39 (2021-05-24): Fix #807 TCP API on ESP32. thanks @jfirwin your hint made the problem easy to find!