meshtastic/firmware #817: Connection Reset By Peer Error With Device as AP

lmatte7, 2021-06-28:
**Describe the bug**
I've been working on adding TCP communications to the Go CLI and I think there may be an issue with the device when it's operating in AP mode. Anytime I try to read from the device (with the Python CLI or my code) I get a `connection reset by peer` error. 
```
meshtastic --host 192.168.42.1 --info                                                                                                                                                                                                                   

ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 54] Connection reset by peer
Traceback (most recent call last):
  File "/usr/local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__main__.py", line 663, in main
    common()
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__main__.py", line 505, in common
    client = TCPInterface(
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__init__.py", line 983, in __init__
    StreamInterface.__init__(
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__init__.py", line 773, in __init__
    self.connect()
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__init__.py", line 795, in connect
    self._waitConnected()
  File "/usr/local/lib/python3.9/site-packages/meshtastic/__init__.py", line 424, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
```

I'm using ` "firmwareVersion": "1.2.39.06892c4"` with the T-BEAM 1.1. 

**To Reproduce**
Set a device to AP mode with a network name and password. 
Connect to the device's wifi network
Use the python CLI to request information from the device 

**Expected behavior**
The device info to be displayed

**Actual behavior**
Commands are sent to the device without issue but a `connection was reset by peer` error occurs when reading data back. 

**Device info:**
 - Device model: T-BEAM 1.1
 - Software Version 1.2.39.06892c4

**CLI Client**
Mac OS X

**Additional context**
The commands work fine when the client and device are both members of the same wifi network

geeksville, 2021-08-02:
oh - I think this was fixed about two weeks ago.  (Someone else reported this in a different issue).  @lmatte7 can you confirm it is now working for you?

lmatte7, 2021-08-06:
I just checked with the latest release (`1.2.43 alpha`) and it works great! Closing the issue

closed 2021-08-06 by lmatte7