meshtastic/firmware #821: t-echo CLI

Justin4462, 2021-07-16:
**Describe the bug**
Unable to access t-echo via CLI

**To Reproduce**
Was unable to install Meshtatic via CLI.
Double pressed reset button to enter developermode.
Installed Meshtastic version 1.2.42 alpha via copy and paste to bootloader method. Install was succesful.
Still unable access device via CLI.

**Expected behavior**
CLI working to configure device 

**Device info:**
 - Device model: T-ECHO
 - Software Version 1.2.42 alpha

24601, 2021-07-23:
Meshtastic CLI is connecting to my T-ECHO, same firmware and update process as your attempt, what version of the CLI are you using, though?

geeksville, 2021-08-02:
hmm - I think this is working and haven't heard back.  Closing for now to keep queue tidy, please reopen if still an issue.

closed 2021-08-02 by geeksville