meshtastic/firmware #825: Unable to Factory Reset RAK4630 Boards

KevinKW87, 2021-07-28:
**Describe the bug**

When trying to reset a RAK4630 board using the python command `meshtastic --set factory_reset true` results in:

ubuntu@ubuntu:~$ meshtastic --set factory_reset true
Traceback (most recent call last):
  File "/usr/local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__main__.py", line 663, in main
    common()
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__main__.py", line 508, in common
    client = SerialInterface(
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 936, in __init__
    StreamInterface.__init__(
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 773, in __init__
    self.connect()
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 795, in connect
    self._waitConnected()
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/__init__.py", line 424, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion

**To Reproduce**
Steps to reproduce the behavior:
1. Run the factory reset python command

**Expected behavior**
Expected to reset device. (Works fine on a T-Beam)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: RAK4630
 - Software Version 1.2.43

geeksville, 2021-08-01:
ok thanks - I'll put this as number two in my queue.

geeksville, 2021-08-01:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/geeksvilles-current-task-list/514/57

garthvh, 2021-12-12:
This works fine, can be closed.

closed 2021-12-12 by sachaw