meshtastic/firmware #826: "Screen.h:290 error: 'SH1106Wire' does not name a type" error when enabling USE_SH1106 in config

dmitryelj, 2021-07-28:
**Describe the bug**
When enabling the USE_SH1106 define in configuration.h we get the compiler error:

```src/esp32/../graphics/Screen.h:290:5: error: 'SH1106Wire' does not name a type
     SH1106Wire dispdev;
```

**Solution**
There is no link from Screen.h to configuration.h. Proper #include should be added at line 23 to Screeen,h:

`#include "../configuration.h"
`

geeksville, 2021-08-01:
Hi @dmitryelj , thank you for this great report.  Alas, I don't have a board with a SH1106 to test with (and my coding time is super limited right now due to some health issues).  Would you be willing to send in a small pull-request with this change?

dmitryelj, 2021-08-01:
Pull request created

dmitryelj, 2021-08-01:
By the way, this issue is duplicated also here: https://github.com/meshtastic/Meshtastic-device/issues/797

closed 2021-08-01 by geeksville

kylegordon, 2021-08-03:
Works great, thank you both :+1:

PR #830 (2021-08-01): USE_SH1106 compilation error fix
Fix for USE_SH1106 compilation error

Fixes #826