meshtastic/firmware #827: RAK46xx Deep Sleeps Inappopriately

HookdomPonix, 2021-07-28:
**Describe the bug**
RAK46xx boards left out on solar enter a deep sleep state without regards to their battery/charging status.  It may also do this regardless of power source.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on RAK46xx
2. Leave on Solar power for unknown amount of time.
3. Board will be in deep sleep (no lights, bt connectivity, or functionality)

**Expected behavior**
Board should not enter this level of sleep at all AFAIK due to no sleep state being necessary for RAK46xx

**Device info:**
 - Device model: RAK4631 Solar Kit
 - Software Version 1.2.43

**Additional context**
When the board is powered again on my 2 test units, they indicated battery percentages of 84pct (unit in strong sunlit position) and 54% (weaker sunlit position).

This bug is also reported by discourse user @claesg.

claesg, 2021-07-29:
Just leave it connected to serial monitor and it will go to deep sleep in 1 to 3 hours. It will close the serial line when it goes to sleep.

IRainman, 2021-07-29:
I have same trouble. T-Echo on NRF52 worked very bad with sleep, it's not connected to the phone on battery and no any buttons worked. Problem is in both of NRF firmware.

geeksville, 2021-08-01:
ooh thanks ya'll.  It sounds like this is the biggest issue for me to work on so I'll start on it today.

geeksville, 2021-08-01:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/geeksvilles-current-task-list/514/57

claesg, 2021-08-05:
Here is a crash log from a RAK4631 with solarpanel and battery attached:
```
13:42:33 7481 [Router] Found existing packet record for fr=0xbff086ac,to=0xffffffff,id=0xd236605
13:42:33 7481 [Router] Add packet record (id=0x0d236605 Fr0xac To0xff, WantAck0, HopLim2 Ch0xb1 encrypted rxSNR=7.25)
13:42:33 7481 [Router] Ignoring incoming msg, because we've already seen it (id=0x0d236605 Fr0xac To0xff, WantAck0, HopLim2 Ch0xb1 encrypted rxSNR=7.25)
13:42:33 7481 [Router] Incoming message was filtered 0xbff086ac
13:42:34 7482 [Power] Battery: usbPower=0, isCharging=0, batMv=4205, batPct=100
Transition powerFSM transition=mesh timeout, from=DARK to=SDS
13:42:36 7484 [PowerFSM] Entering deep sleep for 31536000 seconds
13:42:36 7484 [PowerFSM] Disable NRF52 bluetooth
13:42:36 7484 [PowerFSM] GPS prepare sleep!
13:42:36 7484 [PowerFSM] GPS deep sleep!
13:42:36 7484 [PowerFSM] WANT GPS=0
13:42:36 7484 [PowerFSM] sx1262 entering sleep mode (FIXME, don't keep config)
```
It is running device FW 1.2.43

HookdomPonix, 2021-08-05:
I see something interesting on mine - it looks like power management is in flux potentially:

17:48:58 1702 [Power] Battery: usbPower=0, isCharging=0, batMv=4209, batPct=100
17:49:18 1722 [Power] Battery: usbPower=0, isCharging=0, batMv=4209, batPct=100
17:49:38 1742 [Power] Battery: usbPower=1, isCharging=1, batMv=4215, batPct=0
17:49:58 1762 [Power] Battery: usbPower=1, isCharging=1, batMv=4215, batPct=0

USB was connected for this whole endeavour.

claesg, 2021-08-06:
I think I have found the problem.
With this workaround I have had the RAK4631 running all night. 
Code is at the end of PowerFSM.cpp
Give meshSds different values depending on boards.
```
    auto meshSds = 0;

#ifndef NRF52_SERIES
    // We never enter light-sleep or NB states on NRF52 (because the CPU uses so little power normally)

    // I don't think this transition is correct, turning off for now - @geeksville
    // powerFSM.add_timed_transition(&stateDARK, &stateNB, getPref_phone_timeout_secs() * 1000, NULL, "Phone timeout");

    powerFSM.add_timed_transition(&stateNB, &stateLS, getPref_min_wake_secs() * 1000, NULL, "Min wake timeout");
    powerFSM.add_timed_transition(&stateDARK, &stateLS, getPref_wait_bluetooth_secs() * 1000, NULL, "Bluetooth timeout");
    meshSds = getPref_mesh_sds_timeout_secs();
#else
    lowPowerState = &stateDARK;
    meshSds = UINT32_MAX; //Workaround for now: Don't go into deep sleep on the RAK4631
#endif

    if (meshSds != UINT32_MAX)
        powerFSM.add_timed_transition(lowPowerState, &stateSDS, meshSds * 1000, NULL, "mesh timeout");
```
It is the last line that makes the board go to deep sleep after ~7000 seconds so we just skip it for the NRF52 boards.
@geeksville Maybe you have a better change that will fix the whole power functionality for the board?
or do you want me to send a pull request?

claesg, 2021-08-06:
> I see something interesting on mine - it looks like power management is in flux potentially:
>
> 17:48:58 1702 [Power] Battery: usbPower=0, isCharging=0, batMv=4209, batPct=100
> 17:49:18 1722 [Power] [...]

I am not sure but if the battery goes above 4210 it seems to switch to 0%
and the other two seems to be dialed in for heltecs without battery.
I found this in Power.cpp
```
    /// For heltecs with no battery connected, the measured voltage is 2204, so raising to 2230 from 2100
    const float fullVolt = 4200, emptyVolt = 3270, chargingVolt = 4210, noBatVolt = 2230
```;

claesg, 2021-08-08:
In 
`getBattPercentage()`
in Power.cpp
there is a return 0 if the voltage gets above the chargingvoltage mentioned above
If i add an 'ifndef' around that line it seems to work above 4210:
```
#ifndef NRF52_SERIES
        // This does not work on a RAK4631 with battery connected
        if (v > chargingVolt)
            return 0; // While charging we can't report % full on the battery
#endif
```
I think the return value from this function should be clamped to be between 0 and 100.
```
11:12:39 8482 [Power] Battery: usbPower=1, isCharging=1, batMv=4214, batPct=100
11:12:59 8502 [Power] Battery: usbPower=0, isCharging=0, batMv=4209, batPct=100
```

HookdomPonix, 2021-08-08:
Very good find and I suspected as much.  Before your original battery mod in .43 values were way above possible voltage due to incorrect divider configuration.  

A few things:   
Its possible (but not for certain) that when battery percentage = 0 that it may trigger some of the responses in the state machine around a thoroughly drained battery.  
As such I recommend we set the batPct to 100 when charging to avoid that potential issue.  
-----

Along the lines of that recommendation, I would recommend we implement a battery value smoothing algorithm so that if we get a bogus report from the nrf52 due to a load issue we don't immediately panic and drop into deep sleep.  

Pull the master battery value from 3 polling cycles for example.  That would give us a 60 second average, and we would discard returns greater than 50% away from the previous value for a few cycles as that is implausible barring battery failure (or changing an 18650).  If the battery value change is out of bounds for 5 cycles we begin averaging it into the previous value. 

------

Is it possible it might be smart to restructure the codebase to change from negative declarations of intent based on variant to core functions split by variant within positive declaration blocks?  

It does this some already, but it looks like PowerFSM and Power are two subsystems getting a lot of attention right now and a positive declaration model might help deconflict those work streams.

I'm trying not to backseat drive on this situation and get my stuff setup to contribute fully - just working my way towards a working build enviro for the device and the android app as I have some ideas there too.

claesg, 2021-08-08:
I put it outdoors to see how the solar panel does with 4-5 hours of sun per day and to see if the fix for voltage percentage does help with the sleeping problem. 
It did go down at 0400 this morning and the last report I got from it was 89% so there was no risk of getting 0% due to this 4210 setting. We'll see how it does tomorrow morning.

@HookdomPonix maybe you should put the improvement for the analogue voltage measurement into an Issue/improvement so it does not get lost?

geeksville, 2021-08-09:
I think HookdomPonix's fix is the cleanest proposal.  If while charging, we
should report current % full as 100% rather than 0%.  Alas I'm super
overbooked right now.  If one of ya'll with a RAK could test and send in a
pull request it would be super appreciated.

claesg, 2021-08-09:
I have had it running for 28 hours now outdoors. It reported 95% when I started and now it is down to 77%. We have had a cloudy day today.
I'll send in a pull request first thing tomorrow morning with my changes and then we have something that is at least working (but not as nice as @HookdomPonix suggests)

geeksville, 2021-08-09:
cool beans!  also - I bet this explains the techo shutdowns as well (I was
unable to investigate/program this last weekend).  So if you could try to
structure your patch so it also would be applied on that platform it would
be ideal.  Thanks so much for your help/leadership!

claesg, 2021-08-10:
My RAK4631 has now been running for 42 hours.
I sent in the PR above and tried to include stuff for the T-Echo.

geeksville, 2021-08-11:
Sounds good.  I'll merge it today.  Also I recently updated our nrf52s
Arduino fork from master and I think that change broke USB.  I'll fix that
as well.

(Sent from a phone - please ignore typos)

HookdomPonix, 2021-08-11:
Doing some more research (may have mentioned this already but I don't see it), the TinyUSB 1.4.0 library that is used in the builds before had a race condition issue with the NRF52's that may have been causing some of the negative behaviours.  

Upgrading to 1.4.2 fixes, supposedly.  I'll be testing once USB is back up and claesg's stuff gets merged.  

Thanks for your hard work @geeksville, looking forward to contributing more.

claesg, 2021-08-12:
It ran for 3 days and 9 hours, from 97% down to 55%. Then it dropped off the net and stopped reporting.
A quick push on the reset button brought it back and it continued to run from 54%.
I think there is still something that is not correct in the FW but this fix is at least a step in the right direction.
My RAK4631 is in a RAK solar box and has a recycled 18650 with 2300 mAh inside.

HookdomPonix, 2021-08-12:
Thank you for your testing.   

My earlier concern about averaging the battery percentages to deal with a temporary blip in the was due to timed FSM transitions not being the only possible trigger of the SDS state.  

    // Handle critically low power battery by forcing deep sleep
    powerFSM.add_transition(&stateBOOT, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");
    powerFSM.add_transition(&stateLS, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");
    powerFSM.add_transition(&stateNB, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");
    powerFSM.add_transition(&stateDARK, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");
    powerFSM.add_transition(&stateON, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");
    powerFSM.add_transition(&stateSERIAL, &stateSDS, EVENT_LOW_BATTERY, NULL, "LowBat");

It appears we also trigger SDS when battery percentage drops below a certain percentage to avoid damage to the LIPO type cells as well as ensure that things are written to the flash.  

My concern is that a spurious event of battery = 0 would cause the SDS transition.  

Possible testing approach =  plug in a USB power pack (one that doesn't sleep automatically when it is under low current draw) and see if the USB pull up prevents this behavior.

garthvh, 2021-12-29:
This should be fixed, are you still seeing this issue @HookdomPonix

claesg, 2021-12-29:
After this fix I added one more thing in #871
I noticed that the voltage measurement was a bit flaky so I added a counter for low voltage readings. We now need more than 3 low readings before we change state to EVENT_LOW_BATTERY.

mc-hamster, 2021-12-29:
hysteresis! brilliant!

garthvh, 2022-01-21:
Can we close this?

closed 2022-01-21 by thebentern