meshtastic/firmware #828: RAK4631 loses "owner name" upon powering down

IZ1IVA, 2021-07-30:
**Describe the bug**
After the unit has been powered down, either by the user or on itself by battery depletion and subsequently powered back up, its "owner name" resets to the default "Unknown" + last four digits of its "id".

**Device info:**
 - Device model: RAK4631
 - Software Version: 1.2.43a

geeksville, 2021-08-01:
ooh it sounds like we are not calling flash write after setting prefs from the API.  Will put as #3 in my queue

geeksville, 2021-08-01:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/geeksvilles-current-task-list/514/57

IZ1IVA, 2021-08-02:
Actually it has just happened also on a T-Beam running 1.2.42!

IRainman, 2021-08-04:
Same problem on version 1.2.43. Problem is in both of NRF firmware and that's great realy :) 

> 
> 
> Actually it has just happened also on a T-Beam running 1.2.42!

thebentern, 2021-12-29:
Is this issue still occurring for anyone on the latest release? I haven't experienced it on my RAK or T-Beam devices

garthvh, 2022-01-01:
Nope

thebentern, 2022-01-01:
Closing. If anyone is still experiencing this on the latest software, please open a new issue with the details.

closed 2022-01-01 by thebentern