meshtastic/firmware #834: The meassage queue needs to be rewrited because the messages transmitted to the channel in in unordered

IRainman, 2021-08-07:
I use T-Echo with version 1.2.43. I see the history of the last commit and probably problem is happend after this commit: https://github.com/meshtastic/Meshtastic-device/commit/986d44873a464fe97d09a26260f55e7e7f3f106c

We realy need to use std::queue for the queue ;) std::unordred_set is unusable here! Please don't broke the functionality! Ordering of message is realy need to be same as message are realy sended 

[image]
[image]

garthvh, 2022-01-22:
This is working properly on all of my devices and we don't have any other reports of this issue, can we close this?

closed 2022-01-22 by mc-hamster

garthvh, 2023-02-13:
> i have this bug too, please help is not going order

There are potentially delays for transmitting and receiving, order is not guaranteed, this is not a bug

IRainman, 2023-02-14:
> > i have this bug too, please help is not going order
> 
> There are potentially delays for transmitting and receiving, order is not guaranteed, this is not a bug

This is unusual at all. The protocol is needs to be updated.