meshtastic/firmware #835: Firmaware needs to be process and display text in the utf8 codeing.

IRainman, 2021-08-07:
Currently I use version 1.2.43 and any other language isn't working fine. I send two messages first is:
привет! :)
and second:
hello! ;)
The first message isn't processing correctly

App processing message correctly and text is readable in the app but not on device display. 

[image]

[image]
[image]
[image]
[image]

geeksville, 2021-08-09:
alas - on the device we only have ASCII characters in our font.   So for
any other utf8 characters  intended to show them as question mark.
Apparently that isn't happening.  Adding 'on-device' support for bigger
fonts would depend on someone adding those fonts (which is kinda a big task)

IRainman, 2021-08-11:
Thanks. I understand.