meshtastic/firmware #836: T-Echo does not keep stable gps fix

feh123, 2021-08-08:
**Describe the bug**
I have two T-Echo's, after installation they initially get a gps fix but after a few minutes (<5) the sateillite number falls to zero and then are never updated. If you hike with them there is no change in gps position.

**To Reproduce**
Steps to reproduce the behavior:
1. Reset or start device.
2. Go to satellite page
3 Watch satellite icon.
4 Try hiking with them.

**Expected behavior**
Expect to see several satellites (>10) and distances to base tbeam at home and location lat/lon change.

**Screenshots**
See attached photo of satellite page. I am not sure what all the small icons before the 0 figure mean.  

**Device info:**
 - Device model: LILYGO® T-Echo NRF52840 SX1262 433 / 868 / 915MHz Module LORA GPS 1.54 E-Paper BLE NFC for Arduino. 868Mhz version.
 - Software Version: Can use 1.2.42.or 1.2.43 from meshtastic release page or uf2 from https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo. The version on the Xinyuan site seems to be the best as it has a full set of working button actions but it has the same gps issue. I have tried region set and unset. 

**Smartphone information (if relevant):**
Not relevant
**Additional context**
Sleep issue present with all software options. This bug has been descirbed already.
[image]
[image]
[Not sure this photo will show up so I have tried two versions!]

andrekir, 2021-10-16:
@feh123 seems resolved on firmware 1.2.46a, can you confirm?

feh123, 2021-10-17:
Hi @andrekir thanks for checking with me. I have installed the new version on both T-Echos. There is progress - they both respond in powered mode. However in unpowered mode both buttons are still inactive.
I now have an iOS app thanks to garth and in powered mode I can connect to my T-Echos. Messaging them does not work. Both T-Echos indicate no gps lock and no sats yet my mesh map on my phone shows them correctly positioned.

andrekir, 2021-12-03:
@feh123 can you confirm GPS fix is working as expected for your T-Echo with firmware 1.2.47?

feh123, 2021-12-03:
Hi @andrekir I have been trying both 1.2.47 and 48 - 47 seems best. With is_always_powered I do get the buttons to remain active after I disconnect the usb power. I do see satellites and a gps fix (I changed the antenna) - but it is often wrong. It does receive messages now from the iOS app although the radio node does not appear in the app! Some clear bugs in the gps and the radio still.

feh123, 2021-12-04:
Hi @andrekir I should add that after a few hours (certainly if left overnight) all the buttons are inactive even with is_always_powered true. You need to power the device from a usb and press reset!

closed 2021-12-12 by sachaw