meshtastic/firmware #838: nrf52 targets stall reading from pseduo serial device sometimes

geeksville, 2021-08-13:
while merging #837 I noticed the following:

Sometimes nrf52 targets will stall reading from the pseduo serial port.  This is only apparent if you have the port open via a terminal program or "meshtastic --noproto".

Steps to repro:
1. Run "screen /dev/ttyACM0 921600"
2. See a bunch of debug output
3. After about an hour (on average), you'll see the status LED on the tbeam stop blinking.  
4. Blinking stops until you send the device a character on the serial port.  Which implies the polling on the pseduo device is blocking when it should not.

geeksville, 2021-08-13:
hmm the every 20 second battery messages are not getting printed promptly.  they seem to sit in a queue until a char is sent from the pc

geeksville, 2021-08-13:
the problem is occurring below the level of Stream.flush(), it is in the nrf52 usb code?

HookdomPonix, 2021-08-13:
Tiny USB 1.4.2 loaded in your enviro?  Or 1.4.0?  

1.4.0 had a race condition on the NRF52.

geeksville, 2021-08-15:
1.4.3 alas - I think they still have a problem.  I'll work on a fix soon but my back is bugging me so it will probably be another couple of days...

geeksville, 2021-08-15:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/geeksvilles-current-task-list-now-working-on-nrf52-fixes/514/66

geeksville, 2021-08-15:
OH!  I think I just found the bug.  Testing now - looks good!  USB client code on linux hangs because it seems to expect to always receive a response to reads from the device (even if no characters are pending).  Without that response the cdc driver on linux hangs eventually (until a new character is sent from linux to the device)

```

    if ( 0 == tud_cdc_n_write_flush(itf) )
    {
      // If there is no data left, a ZLP should be sent if
      // xferred_bytes is multiple of EP Packet size and not zero
      if ( !tu_fifo_count(&p_cdc->tx_ff) && xferred_bytes /* && (0 == (xferred_bytes & (BULK_PACKET_SIZE-1))) */ )
      {
        if ( usbd_edpt_claim(rhport, p_cdc->ep_in) )
        {
          usbd_edpt_xfer(rhport, p_cdc->ep_in, NULL, 0);
        }
      }
    }
```

geeksville, 2021-08-15:
fixed in Adafruit_TinyUSB_Arduino.  The next clean build (for anyone who does "pio update") should get this fix automatically.

geeksville, 2021-08-15:
fixed in lib above

closed 2021-08-15 by geeksville