meshtastic/firmware #846: T-Echo bugs

andrekir, 2021-08-30:
From Meshtastic forum post: https://meshtastic.discourse.group/t/t-echo-freezing-when-loosing-bluetooth/3786

- Device model: TTGO T-Echo
- Firmware tested 1.2.44 & 1.2.45

1. reset button doesn't work on battery power (works fine when connected to USB power);

2. freezing when bluetooth disconnects on battery power (works fine when connected to USB power);
(thanks @apt105 for finding root cause)

2.1. Connect T-Echo to Android App via bluetooth;
2.2. Disable bluetooth on Android;
2.3. T-Echo flashes red LED twice;
2.4. T-Echo no longer responds, no LEDs or bluetooth activity.

3. GPS sometimes flips the signs (+/-) on the coordinates (-16.696168, -50.359996 → +16.696168, +50.359996);

[image]

4. main screen with GPS satellite locks will only display correctly at power on, later always reads “0”.

andrekir, 2021-08-30:
Item 4 is the same as: https://github.com/meshtastic/Meshtastic-device/issues/836

andrekir, 2021-10-15:
>     2. freezing when bluetooth disconnects on battery power (works fine when connected to USB power);
Firmware release 1.2.46a seems to have solved this issue.

andrekir, 2021-10-16:
@ducky64, can you confirm you have 1.2.46a? is there a pattern for the freeze to occur? any steps to reproduce it?

> A reset by the top button is necessary, and the reset still only works when connected to USB power.

same for me.

> I may also just be using it wrong: the advanced settings text isn't super descriptive either - though I'd hope it shouldn't configure the device into a non-responsive state?

there shouldn't be a wrong way of using the advanced settings, but... are you using "Device sleep period = 0"? I could try that. default value is 300, just in case.

andrekir, 2021-10-16:
after more testing, also consistently getting stuck in sleep after some time. steps to reproduce are different now:

- for me only happened on battery power (not on USB power);
- bluetooth connection & battery percentage doesn't seem to matter (from 80% to 50%);
- plugging USB cable seems to bring back from sleep (doesn't require reset, like before).

behaves a lot as if the device had run out of battery.

ducky64, 2021-10-16:
Yes, on 1.2.46a, version on the Android app shows up as `1.2.46-dce2fe4`

I had device sleep period set to 30, and after setting that to 0 (which seemed to default back to 300 after re-connecting to the radio), the device no longer seems to freeze (where the buttons don't respond or the blue LED stops blinking - so far tested over a few hours). However, after some time (not sure how much yet), text messages no longer send from the app (they don't show up on the other device's screen or the other phone). Resetting both radios (via the reset button) fixes this.

ducky64, 2021-10-16:
... and looks like one just went into a dead sleep. It's been sitting unplugged for under a half hour. Upon reset it indicates 4.11V, 91%

geeksville, 2021-10-21:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-device-alpha-1-2-46-is-ready-for-alpha-testers-includes-t-echo-and-rak-fixes/3969/17

Miskerest, 2021-11-11:
I just tried replicating this bug and also observed this weirdness:

```
meshtastic --nodes --port COM6
Connected to radio
╒═════╤══════════════╤═══════╤═══════════╤════════════╤═════════════╤════════════╤═══════════╤═════════╤═════════════════════╤══════════════╕
│   N │ User         │ AKA   │ ID        │ Latitude   │ Longitude   │ Altitude   │ Battery   │ SNR     │ LastHeard           │ Since        │
╞═════╪══════════════╪═══════╪═══════════╪════════════╪═════════════╪════════════╪═══════════╪═════════╪═════════════════════╪══════════════╡
│   1 │ Unknown f87b │ ?7B   │ !eca2f87b │ [plz no]   │ [doxerino]  │ 143 m      │ 98.00%    │ N/A     │ 1999-11-29 20:09:36 │ 21 years ago │
├─────┼──────────────┼───────┼───────────┼────────────┼─────────────┼────────────┼───────────┼─────────┼─────────────────────┼──────────────┤
│   3 │ Unknown 9e96 │ ?96   │ !7c269e96 │ N/A        │ N/A         │ N/A        │ 100.00%   │ 4.75 dB │ 1999-11-29 20:08:33 │ 21 years ago │
╘═════╧══════════════╧═══════╧═══════════╧════════════╧═════════════╧════════════╧═══════════╧═════════╧═════════════════════╧══════════════╛
```

[image]

It seems that some timestamps are also getting wiped out/overflowed

runestone74, 2021-11-13:
I tried Python CLI:
meshtastic --set-time
meshtastic.exe: error: unrecognized arguments: --set-time
And the command --no-time don't seem to prevent the T-Echo sending bad time to the timestamp log (in the iOS app) so the messages appear in the incorrect order. Using FW Alpha .46

runestone74, 2021-11-15:
In the code it seems the non-working reset button on battery was introduced in 1.2.43. These are the differences in \src\nrf52\main-nrf52.cpp

v 1.2.42
bool loopCanSleep() {
    return !tud_cdc_connected();
}

v 1.2.43
bool loopCanSleep() {
    // turn off sleep only while connected via USB
    // return true;
    return !Serial; // the bool operator on the nrf52 serial class returns true if connected to a PC currently
    // return !(TinyUSBDevice.mounted() && !TinyUSBDevice.suspended());
}

Can anyone confirm this? Why was this changed in the first place?

Miskerest, 2021-11-15:
That's where I was looking too. I'm wondering why this wouldn't affect other nrf devices?

Miskerest, 2021-11-15:
It looks like the esp32 code just returns true:

main.cpp:
```
/**
 * Some platforms (nrf52) might provide an alterate version that supresses calling delay from sleep.
 */
__attribute__ ((weak, noinline)) bool loopCanSleep() {
    return true;
}
```

runestone74, 2021-11-16:
So would it be safe to assume if the above is changed back to the "working" condition in main-nrf52.cpp, it would be a permanent fix? Who can make this fix? I am not comfortable to to the github magic yet...

andrekir, 2021-11-16:
@runestone74 just so we are on the same page. do you mean fw 1.2.42 the reset button works for you without usb power? because it does not for myself.

and make sure you are using official releases from Meshtastic GitHub. the only fw the reset button works for me is the LilyGO GitHub build version "t-echo 1.2.42.753ae00-d" (different from Meshtastic release 1.2.42.2759c8d).

runestone74, 2021-11-16:
You are correct, I was comparing 1.2.46 with the 1.2.42.2759c8d release, not the 753ae00-d.

runestone74, 2021-11-16:
So where can the source for "t-echo 1.2.42.753ae00-d" release be found?

sachaw, 2021-12-12:
@andrekir Can we please get an update of the status of individual issues still present so we can track them, thanks.

andrekir, 2021-12-12:
@sachaw from my tests up to firmware 1.2.47, still present:

1. reset button doesn't work on battery power (works fine when connected to USB power but not with `is_always_powered` true);
2. freezing when bluetooth disconnects (works "better" when connected to USB power but not with `is_always_powered` true);
3. GPS sometimes flips the signs (+/-) on the coordinates (-16.696168, -50.359996 → +16.696168, +50.359996).

sachaw, 2021-12-12:
Closed in favor of specific issues linked above.

closed 2021-12-12 by sachaw