meshtastic/firmware #852: T-echo goes into permanent sleep on battery power and never wakes

airjer18, 2021-09-02:
**Describe the bug**
T-echo goes into permanent sleep on battery power and never wakes, unless plugged into a computer via USB

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.2.45 firmware for techo
2. Unplug from usb
3. Watch it initially function for 20 secs and go into a permanent sleep
4. Absolutely no way to get it to function at all on battery power, tried removing battery plugging back into usb and as long as it's plugged into usb it will work, plug battery back in, and disconnect from USB, about 20 secs, then it goes to sleep
5. Tried setting Device sleep period to 0, to no effect, default setting keeps popping up with 86400
6. Cannot get Techo to respond at all on battery power  only USB into a computer, it won't react when connected to battery bank
7. tried rolling back firmware to 1.2.42/43/44   Same problem,   Tried rolling back to 1.2.39 and the techo won't even restart to load the firmware.

**Expected behavior**
Expected for it to function on battery power

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. T-ECHO]
 - Software Version [e.g. ALL]

datagod, 2021-09-02:
This same problem is occurring on my TTGO T-Beam.  Two of the devices are now bricked.

andrekir, 2021-09-07:
Duplicate of #846

claesg, 2021-09-12:
The same problem exists on the RAK4631. It will go 'offline' after a few days and needs power or reset button push to come back. 
I suspect the voltage reading is a bit flaky and one reading that is below 10% will send it to deep sleep
I will add a counter that increments on low voltage readings and decrements on ok readings.
If we get more than 3 low readings in a row the we go to sleep.
I'll do some testing to see how long it will run with this change.

HookdomPonix, 2021-09-14:
I can also confirm this behaviour on the RAK4631 still exists with .45.

geeksville, 2021-09-15:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/t-echos-reliability/3817/9

claesg, 2021-09-15:
Now my test has been running for 3 days and it is down to 62% on the battery reading. This is where it starts to get interesting. On the unmodified code it usually went offline about now. 
The RAK4631 is not connected to a phone and there ar 4 more nodes in the house just doing location updates. I have one node connected to a computer which is doing 'meshtastic --nodes' every 20 minutes and storing in a file to keep track of things.

freedomtowander-code, 2026-02-17:
My lilygo t deck does this as well

goldentg, 2026-04-09:
having this problem with my T-Echo as well