meshtastic/firmware #860: Can not send yet, busyTx

tomasbrincil, 2021-09-08:
Hi there, I am using T-Beam v 1.1 sx1262, I am trying to desolder original chip a and wire G-NiceRF sx1262 module with integrated power amplifier (I am licenced ham I should know what I am doing :))

[image]

Considering same wiring for G-NiceRF module as original module there is some issue when trying to transmit.

> [15:40:40]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [15:40:40]configsip: 0, SPIWP:0xee
> [15:40:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> [15:40:40]mode:DIO, clock div:2
> [15:40:40]load:0x3fff0018,len:4
> [15:40:40]load:0x3fff001c,len:1044
> [15:40:40]load:0x40078000,len:8896
> [15:40:40]load:0x40080400,len:5828
> [15:40:40]entry 0x400806ac
> [15:40:41][D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
> [15:40:41][15:40:41]??:??:?? 0 Filesystem files:
> [15:40:41]??:??:?? 0   /prefs/db.proto
> [15:40:41]??:??:?? 0   /prefs/radio.proto
> [15:40:41]??:??:?? 0   /prefs/channels.proto
> [15:40:41]??:??:?? 0 I2C device found at address 0x34
> [15:40:41]??:??:?? 0 axp192 PMU found
> [15:40:41]??:??:?? 0 I2C device found at address 0x3c
> [15:40:41]??:??:?? 0 ssd1306 display found
> [15:40:41]??:??:?? 0 done
> [15:40:41]??:??:?? 0 Meshtastic hwvendor=4, swver=1.2.45.a27ebda-d, hwver=unset
> [15:40:41]??:??:?? 0 Setting random seed 293909529
> [15:40:41]??:??:?? 0 Total heap: 254984
> [15:40:41]??:??:?? 0 Free heap: 221212
> [15:40:41]??:??:?? 0 Total PSRAM: 4194252
> [15:40:41]??:??:?? 0 Free PSRAM: 4194252
> [15:40:41]??:??:?? 0 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630
> [15:40:41]??:??:?? 0 Setup Preferences in Flash Storage
> [15:40:41]??:??:?? 0 Number of Device Reboots: 29
> [15:40:41]??:??:?? 0 Initial packet id 1718671776, numPacketId 4294967295
> [15:40:41]??:??:?? 0 Setting default channel and radio preferences!
> [15:40:41]??:??:?? 0 Expanding short PSK #1
> [15:40:41]??:??:?? 0 Wanted region 0, using Unset
> [15:40:41]??:??:?? 0 Loading /prefs/db.proto
> [15:40:41]??:??:?? 0 Loaded saved preferences version 11
> [15:40:41]??:??:?? 0 Loading /prefs/radio.proto
> [15:40:41]??:??:?? 0 Loading /prefs/channels.proto
> [15:40:41]??:??:?? 0 Number of Device Reboots: 29
> [15:40:41]??:??:?? 0 Expanding short PSK #1
> [15:40:41]??:??:?? 0 Wanted region 0, using Unset
> [15:40:41]??:??:?? 0 region=0, NODENUM=0xf267a2b8, dbsize=1
> [15:40:41]chip id detect 0x3
> [15:40:41]Detect CHIP :AXP192
> [15:40:41]OUTPUT Register 0x5f
> [15:40:41]??:??:?? 0 AXP192 Begin PASS
> [15:40:41]??:??:?? 0 DCDC1: ENABLE
> [15:40:41]??:??:?? 0 DCDC2: ENABLE
> [15:40:41]??:??:?? 0 LDO2: ENABLE
> [15:40:41]??:??:?? 0 LDO3: ENABLE
> [15:40:41]??:??:?? 0 DCDC3: ENABLE
> [15:40:41]??:??:?? 0 Exten: ENABLE
> [15:40:41]??:??:?? 0 ----------------------------------------
> [15:40:41]??:??:?? 0 DCDC1: ENABLE
> [15:40:41]??:??:?? 0 DCDC2: ENABLE
> [15:40:41]??:??:?? 0 LDO2: ENABLE
> [15:40:41]??:??:?? 0 LDO3: ENABLE
> [15:40:41]??:??:?? 0 DCDC3: ENABLE
> [15:40:41]??:??:?? 0 Exten: ENABLE
> [15:40:41]SRC REG:0xc4
> [15:40:41]Charging enable is enable
> [15:40:41]Charging target-voltage : 0x2
> [15:40:41] end when the charge current is lower than 10% of the set value
> [15:40:41]Charge current : 700.00 mA
> [15:40:41]??:??:?? 0 Battery: usbPower=1, isCharging=1, batMv=4187, batPct=100
> [15:40:41]??:??:?? 0 Read RTC time as 408 (cur millis 209) quality=0
[... 158 lines cut ...]
> [15:42:15]??:??:?? 94 [RadioIf] delay done
> [15:42:15]??:??:?? 94 [RadioIf] Can not send yet, busyTx
> [15:42:15]??:??:?? 94 [RadioIf] Hardware Failure! busyTx for more than 60s
> [15:42:15]??:??:?? 94 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:50
> [15:42:15]??:??:?? 94 [RadioIf] xmit timer 1605
> [15:42:15]??:??:?? 94 [Screen] Screen: Critical error 8!
> [15:42:17]??:??:?? 95 [RadioIf] delay done
> [15:42:17]??:??:?? 95 [RadioIf] Can not send yet, busyTx
> [15:42:17]??:??:?? 95 [RadioIf] Hardware Failure! busyTx for more than 60s
> [15:42:17]??:??:?? 95 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:50
> [15:42:17]??:??:?? 95 [RadioIf] xmit timer 3125
> [15:42:17]??:??:?? 95 [Screen] Screen: Critical error 8!
> [15:42:20]??:??:?? 99 [RadioIf] delay done
> [15:42:20]??:??:?? 99 [RadioIf] Can not send yet, busyTx
> [15:42:20]??:??:?? 99 [RadioIf] Hardware Failure! busyTx for more than 60s
> [15:42:20]??:??:?? 99 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:50
> [15:42:20]ets Jul 29 2019 12:21:46
> [15:42:20]
> 

BUSY pin on module is 0V. I tried watch scope and no difference between modules.
Yellow is clock, green is data from master, blue is BUSY.
[image]

Any idea? In your private TODO.md (sorry!) I found "swbug! stuck busy tx occurred!" on eink. Can´t be same problem?

markbirss, 2021-09-21:
@tomasbrincil  Hi, have you connected the busy pin also ? possibly nned same connection as Type 2 here - https://github.com/NanoVHF/Meshtastic-DIY/blob/main/Schematics/Mesh_NiceRF.pdf.

tomasbrincil, 2021-09-21:
Hi, thank for your time. Yes busy pin is connected to original pin 32. I rewired 1:1 all pads to G-Nice RF module.

markbirss, 2021-09-21:
Oh Ok, my RA-02 would not initialise correctly but i had wired to wrong GPIO on my breakout board
It works now 
Im using the Tlora32 v1 firmware but custom DIY board ESP32 and RA02

[image]

tomasbrincil, 2021-09-23:
No change. I am trying differents GPIO pins and nothing helps. @geeksville can you hint me pls? Is it software thing?

osmanovv, 2021-09-29:
There is no Tx pin issue, "busyTx" in this message means you are still sending a message or forget to handle it:

https://github.com/meshtastic/Meshtastic-device/blob/b182819aff7aece1eaae8efefca749c3384df2aa/src/mesh/RadioLibInterface.cpp#L65-L78

Use https://github.com/meshtastic/Meshtastic-device/blob/b182819aff7aece1eaae8efefca749c3384df2aa/src/mesh/RadioLibInterface.cpp#L211-L218

or https://github.com/meshtastic/Meshtastic-device/blob/b182819aff7aece1eaae8efefca749c3384df2aa/src/mesh/RadioLibInterface.cpp#L220-L225

to clear the message after sending it.

closed 2021-10-05 by tomasbrincil