meshtastic/firmware #863: NMEA GPS - bogus heading (course) data from TinyGPS++

a-f-G-U-C, 2021-09-09:
Hardware TBEAM T22_V1.1 with UBLOX M8N in forced NMEA mode - but the issue doesn't appear to be specific to this device

On rare occasions, the value returned by TinyGPS++ call `reader.course.value()` is obviously bogus (>360 deg) and **does not match** any of the `GNRMC` lines logged from GPS, but: the actual *digits* in the bogus number sometimes (visually) match those in the longitude value (!!)

I suspect this is a TinyGPS++ parsing bug. Posting here to hopefully assist others and collect more information.

The issue occurs irrespective of TinyGPS Custom Fields being enabled or disabled.

a-f-G-U-C, 2021-09-10:
Update: I have traced the problem to an improper variable initialization bug in TinyGPS++
Since Meshtastic uses its own fork of TinyGPS++, I will raise a PR against our fork first. It's only 3 lines of code :)

sachaw, 2021-12-07:
@a-f-G-U-C Sorry, only just saw this was still open, have merged it, please commit the updated dependency and I can merge it, then close this. Thanks!

a-f-G-U-C, 2021-12-08:
No problem @sachaw , and thanks for your help. I think the issue is now completely fixed, with no further action necessary.

closed 2021-12-08 by a-f-G-U-C