meshtastic/firmware #868: Custom DIY device hardware model identifier based on @NanoVHF schematics

NanoVHF, 2021-09-16:
Hello.
Recently, the problem with the lack of support for c1268 chips was found out. (https://meshtastic.discourse.group/t/sx1268-modules-are-not-yet-supported/3659/4)
A colleague of mine did support and submitted some pull requests with support for the new SX1268 module (https://github.com/meshtastic/Meshtastic-device/pull/865) and reserved a name for the new homemade device (https://github.com/meshtastic/Meshtastic-protobufs/pull/25).

We propose to make a separate section in platforms.ini and configs for self-made devices "Meshtastic-DIY".
It is also planned to introduce support for new CC68 chips for the maximum number of supported LoRa devices.

osmanovv, 2021-09-17:
Link to Meshtastic-DIY project discussion https://meshtastic.discourse.group/t/meshtastic-diy-project/3831

osmanovv, 2021-09-19:
Settings for building it as separate board configuration: #870

Declare `default_envs = meshtastic-diy-v1` in platformio.ini `[platformio]` section.