meshtastic/firmware #912: NodeInfo plugin outputting malformed protobufs

sachaw, 2021-10-30:
various errors occur when validating  protobufs outputted  from the NodeInfo plugin against the schema, here are some example of malformed packets, these can be visualized at https://protobuf-decoder.netlify.app/

Premature EOF:
```
CgkhMzM2YWJkNTESDFVua25vd24gYmQ1MRoDPzUxIgbETzNqvVEwBA==
```
```
CgkhN2RlMzJkZjgSDFVua25vd24gMmRmOBoDP0Y4IgbXf33jLfgwCQ==
```

cant skip wire type 4:
```
CgkhYjMwYmQ1MGQSDFVua25vd24gZDUwZBoDPzBEIgb/dbML1Q0wCQ==
```

illegal tag: field no 12 wire type 6:
```
CgkhNWVkZmY4OTASDFVua25vd24gZjg5MBoDPzkwIgbWdV7f+JAwCQ==
```

I have generated a valid, parsable packet which looks like:

```
CHsSGQoDMTIzEgtEZXZpY2UgTmFtZRoDc2h0MAQaDA0hAAAAFYv///8gZCXmT31hPQAAAEA=
```

From my observations, it appears that the device is sending a `User` message and not a `NodeInfo` message

a-f-G-U-C, 2021-11-02:
> From my observations, it appears that the device is sending a User message and not a NodeInfo message

I can confirm that this is definitely the case at the moment:
https://github.com/meshtastic/Meshtastic-device/blob/e21cc9d47947a530bf3282c64f45ffd91ff0f9e7/src/plugins/NodeInfoPlugin.cpp#L45-L51

Whether or not this is intentional, I don't know; and whether or not this is *a good thing*, is a matter for discussion. 
Personally I think I prefer it this way, since the User messages pack all the relevant information in a smaller payload (because no nesting).

sachaw, 2021-11-02:
Yeah, after working with it a little more (and understanding it now). I think this is the better way of handling it. Likely a rename is in order for the plugin name though.

closed 2021-12-10 by sachaw