meshtastic/firmware #916: BLE off when WIFI on (1.2.45) - So how to turn off WIKI to get BLE back?

hb9eue, 2021-11-06:
Hi

I updated from 1.2.25 to 1.2.45 via Andriod App and immediately found, I could not connect via BLE after the update.

So I went on and did a complete reinstall (spiffs reflash) of 1.2.45 and was able to re-pair Bluetooth.
So I used the meshtastic python app to re-connect my WiFi and set the device into HAM-Mode.

BLE was dead again after this.

Yes, the release notes reveal why. So I was looking for the WIFI off/on switch, to disable WIFI to get BLE back.

But I only find a way to turn the AP mode on and off. Not WIFI itself.

So how can I get BLE back to work?

Will  there be a future fix to have BLE and WIFI Co-Exist?

-Benoit-

AngeLinuX99, 2021-11-14:
Hi, I've had the same problem and can't go back to BLE. My only solution has been to flash the firmware again.

sachaw, 2021-11-14:
A workaround for now, as the python cli won't allow sending empty strings (to disable wifi).
If you flash one of the latest build artifacts and get access to the web ui, you can go to `settings` -> `WiFi` and set the SSID and Password to blank, save and your BLE should work again.
this build should work fine: https://github.com/meshtastic/Meshtastic-device/actions/runs/1449106610

closed 2021-12-07 by sachaw