meshtastic/firmware #920: Fonts use up a lot of storage on device. This needs to be cleaned up.

mc-hamster, 2021-11-18:
Flash Memory analysis shows that fonts use a disproportionally large amount of memory due to duplication.

This needs to be cleaned up to reduce the foot print. Code now takes up 92% of total available space.

[image]

runestone74, 2021-11-25:
What kind of cleaning is needed? High level or low level?

mc-hamster, 2021-11-27:
@runestone74 The font "ArialMT_Plain_10" is being stored 5 times and "ArialMT_Plain_16" is being stored twice. They should be stored once and then reused.

See screenshots from the analysis (I should have also included these earlier, thanks for asking!)

[image]

[image]

[image]

caveman99, 2022-01-18:
this can also be seen from the output.map generated during buildtime. Actually 2 more embeds of 10 and 16 size popped up with the CannedReply Plugin. 

```
output.map: .rodata._ZL16ArialMT_Plain_24
output.map-                0x3f401e88     0x25ab .pio/build/heltec-v1/src/graphics/Screen.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_16
output.map-                0x3f404433     0x13b9 .pio/build/heltec-v1/src/graphics/Screen.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_10
output.map-                0x3f4057ec      0xaab .pio/build/heltec-v1/src/graphics/Screen.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_16
output.map-                0x3f40c3c4     0x13b9 .pio/build/heltec-v1/src/plugins/CannedMessagePlugin.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_10
output.map-                0x3f40d77d      0xaab .pio/build/heltec-v1/src/plugins/CannedMessagePlugin.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_16
output.map-                0x3f40eef8     0x13b9 .pio/build/heltec-v1/src/plugins/esp32/EnvironmentalMeasurementPlugin.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_10
output.map-                0x3f4102b1      0xaab .pio/build/heltec-v1/src/plugins/esp32/EnvironmentalMeasurementPlugin.cpp.o
output.map: .rodata._ZL16ArialMT_Plain_10
output.map-                0x3f411d8c      0xaab .pio/build/heltec-v1/lib0c6/libESP8266_SSD1306.a(OLEDDisplay.cpp.o)
output.map: .rodata._ZL16ArialMT_Plain_10
output.map-                0x3f412847      0xaab .pio/build/heltec-v1/lib0c6/libESP8266_SSD1306.a(OLEDDisplayUi.cpp.o)

```

caveman99, 2022-01-18:
There's something fundamentally wrong here. Even including the SSD1306 lib has the Plain_10 font packed twice. I see we include our own version of that lib, maybe something got broken during customization?

caveman99, 2022-01-21:
I think i have solved this ... T-Beam flash sizes before/after:

Flash: [========= ]  94.2% (used 1729091 bytes from 1835008 bytes)
Flash: [========= ]  93.1% (used 1708595 bytes from 1835008 bytes)

Problem is, to solve this the Screen lib needs to be modified. There's a flaw in the font definition logic. I'll send in a PR there.

closed 2022-01-24 by mkinney

mc-hamster, 2022-01-24:
Thanks @caveman99 !!!!