meshtastic/firmware #930: Crash reset on GPIO with wrong bit mask

KiwiHC16, 2021-11-24:
From python script, send a GPIO with wrong mask and your device will restart.

Context:
- 2 meshtastic device, one is connected with USB (master) and second is remote (!f244c480).
- GPIO channel setup done.

File gpioRead.py
```
import time
import meshtastic
from pubsub import pub
from meshtastic import remote_hardware

def onGPIOreceive(packet, interface):
    hw = packet["decoded"]["remotehw"]
    print(f'Received RemoteHardware typ={hw["typ"]}, gpio_value={hw["gpioValue"]}')

interface = meshtastic.SerialInterface("/dev/ttyUSB0")                 
rhc = remote_hardware.RemoteHardwareClient(interface)

dest = "!f244c480" 

while 1:
    time.sleep(3)
    rhc.readGPIOs(dest, 0xFF)

interface.close()
```

Bimask wrongly set to 0xFF, destination !f244c480 will reboot, 

shell# python3 gpioRead.py 
f244c480 reboot

garthvh, 2022-01-26:
Does this still happen @mkinney?

mkinney, 2022-01-27:
Can confirm it still happens:

```
00:09:41 483 [Router] decoded message (id=0x687795aa Fr0xf0 To0xc0, WantAck1, HopLim3 Ch0x1 Portnum=2 WANTRESP rxtime=1643242181 rxSNR=12.25)
00:09:41 483 [Router] handleReceived(REMOTE) (id=0x687795aa Fr0xf0 To0xc0, WantAck1, HopLim3 Ch0x1 Portnum=2 WANTRESP rxtime=1643242181 rxSNR=12.25)
00:09:41 483 [Router] Plugin 'remotehardware' wantsPacket=1
00:09:41 483 [Router] Received remotehardware from=0xc4fcbef0, id=0x687795aa, portnum=2, payloadlen=5
00:09:41 483 [Router] Received RemoteHardware typ=4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008146d  PS      : 0x00060730  A0      : 0x800ea95e  A1      : 0x3ffd32e0
A2      : 0x00000007  A3      : 0x00000005  A4      : 0x3ffd3310  A5      : 0x3ffd32e0
A6      : 0x00000008  A7      : 0x3ffd327c  A8      : 0x3f42e7dc  A9      : 0xffff0300
A10     : 0xffffffff  A11     : 0x00000064  A12     : 0x08000000  A13     : 0x4a000000
A14     : 0xffffffff  A15     : 0x00000008  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0xffffffff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x4008146d:0x3ffd32e0 0x400ea95b:0x3ffd3300 0x400eab24:0x3ffd3330 0x400ea8de:0x3ffd3380 0x400de42d:0x3ffd33d0 0x400e1a0d:0x3ffd3400 0x400e1add:0x3ffd3420 0x400e1afd:0x3ffd3440 0x400dd93d:0x3ffd3470 0x400d525e:0x3ffd3490 0x400f225d:0x3ffd34b0 0x400dce09:0x3ffd34e0 0x4011a5dd:0x3ffd3500

Rebooting...
```

mc-hamster, 2022-01-27:
I’m not sure if this is something we can easily fix. The intent of this feature is to give access to the low level hardware. That makes this feature an expert tool. It would then be up to the expert not to crash the device. 

If anyone can think of an elegant fix, feel free to reopen this issue otherwise… Closing.

closed 2022-01-27 by mc-hamster