meshtastic/firmware #962: Feature Request/Proposal: Site Survey or Scan to locate nearby nodes

SmugIsh0, 2021-12-06:
Hello, I am new to this project and not very radio literate so please bear with me on this, I was asked to create a feature request by MC Hamster via Discord. it’s a very broad idea and I’m not sure all of it is feasible. The general idea is it would be nice to have a way so scan an area to see if there are any devices nearby so you can possibly contribute to an already existing mesh. Not join their encrypted chats, but match node configuration communications settings to extend the range of the existing mesh.

- A way to do a “site survey” to see if there are any nodes within range

- As a device owner, allow/deny your device to be detected by a “site survey”, for privacy

- As a device owner, allow/deny your device to give out its frequency, spreading factor and coding rate when detected by a “site survey”
          -- I am not aware of what the implications are of giving out this type of information, I’m not sure if it would create a security or 
              privacy issue

Context: This came up while I was asking questions to better understand the way devices interact with each other. I wanted to know if 2 mesh networks (that had no prior knowledge of each other) came in contact with one another, could they utilize each other’s nodes. The answer is that they would have to be on the same frequency, spreading factor and coding rate. Its my understanding that the chances of that happening are low, so I was wondering if there was a “site survey” option to see what devices are nearby, and what their general configuration was.

I understand a lot of people want to have a totally private mesh network and not have drive by nodes and other mesh’s utilizing their nodes, so this bit should be optional. But for those of us that would like to extend an already existing nearby mesh or plant a seed for the start of a bigger mesh, it would be nice to have a way to see what’s in the area and match freq/spreading factor/coding rate so a node’s settings can be matched up.

joshpirihi, 2021-12-06:
One thing I've been thinking about is a traceroute type utility.  This could be a part of a site survey - it could draw a graph of the mesh network

garthvh, 2022-09-12:
This is probably better accomplished in client software, nodes that just do repeating (which is what this request is looking to scan for) are not part of the nodedb.

closed 2022-09-12 by garthvh

mfonhof, 2024-01-20:
What's the status on this?

macattackpro, 2024-02-08:
> What's the status on this?

Closed on Sep 12, 2022. This would be better suited with an SDR and a RPI. That said, there is a project you could resurrect from 7 years ago. https://github.com/ImprobableStudios/Feather_TFT_LoRa_Sniffer