meshtastic/firmware #967: Feature reques

ANS-spb, 2021-12-07:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. TBEAM]
 - Software Version [e.g. 0.7.8]

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 0.7.2]

**Additional context**
Add any other context about the problem here.

geeksville, 2021-12-07:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-meshtastic-python-v1-2-44/4356/4

mc-hamster, 2021-12-07:
This is done.

closed 2021-12-07 by mc-hamster