meshtastic/firmware #970: Feature request: add airtime to NodeInfo

PlumRugOfDoom, 2021-12-07:
Add airtime to NodeInfo, useful for additional features like packet rate limit / duty cycle limit to stay within region specific regulations

mc-hamster, 2021-12-07:
What will go into nodeInfo will be three arrays

- periodTX - 48 time windows of 1hr each with the airtime transmitted out of the device per hour.

- periodRX - 48 time windows of 1hr each with the airtime of valid packets for your mesh.

- periodRX_ALL - 48 time windows of 1hr each with the airtime of valid lora packets. If you subtract periodRX from periodRX_ALL, you'll get something that looks like the measure of noise on the selected frequency.

mc-hamster, 2021-12-09:
this is done

closed 2021-12-09 by mc-hamster