meshtastic/firmware #990: T-Echo - Reset doesn't work on battery

sachaw, 2021-12-12:
From: #846
Reset button doesn't work on battery power (works fine when connected to USB power but not with `is_always_powered` true.

andrekir, 2021-12-12:
the only firmware I know the reset button works is the elusive (and sourceless) LilyGO GitHub build version "t-echo 1.2.42.753ae00-d" (different from Meshtastic release 1.2.42.2759c8d).

sachaw, 2021-12-12:
I'll try contacting them tomorrow with a GPL enforcement email.

mc-hamster, 2021-12-12:
@andrekir @sachaw 

1.2.42.753ae00-d is here:

https://github.com/meshtastic/Meshtastic-device/commit/753ae00ddbe14ae44968b53aca8d86f4fd0a19d4

Kevin did a build from that commit.

andrekir, 2021-12-12:
worth mentioning: pressing the reset button without USB power will cause a freeze/crash (so not just "doesn't work").

bagmanden, 2021-12-20:
also worth mentioning: When putting the t-echo to sleep. then you can not wake it up again. Only by plugging power into it

geeksville, 2022-04-27:
Lily at TTGO just sent this great email.  Alas, I don't have a TECHO with me in here in Taiwan (and I'm trying to avoid standing in front of the computer much).  Can one of ya'll volunteer to try making this change and see if it fixes the problem?

> Through customer feedback, our engineers found that PIN_EINK_PWR_ON needs to be increased in the firmware, such as picture content.  Hope this feedback is useful to you -Lily
[image]

geeksville, 2022-04-27:
see related pull request here: https://github.com/meshtastic/Meshtastic-device/pull/1408 (just cross linking for clarity)

caveman99, 2022-04-27:
the 5 second pause is plausible. I put that in for early message debugging and it occurs if the device is not connected to a serial port. Code was donated by kongduino from RAK.

andrekir, 2022-04-27:
for reference, now this is how the reset behaves on my end after changes in 1.3.8.13fa7c1-d or 1.3.10.cc2a84a:

https://user-images.githubusercontent.com/48680741/165589306-f611d118-6aca-486f-8cae-4e700ee40063.mp4

caveman99, 2022-04-28:
There are reports on discorse that this fix indeed made the t-echo work when not powered. Does that warrant a backport to 1.2?

geeksville, 2022-04-29:
ooh this sounds like a good fix then.  I'm still going to spend some time in the next few days to check the schematics and try to figure out **why** it works ;-).  Good job @Pedestrian11!

I'd vote for a backprop to 1.2 but really the decision is up to you active devs.

mc-hamster, 2022-04-29:
My vote is to also back port this to 1.2. That'll be the gold standard for a few more months and this is a real usability issue now.

andrekir, 2022-05-04:
done! big thanks to everyone involved!

closed 2022-05-04 by andrekir

surlyhacker, 2022-07-09:
I guess a component of this bug is that the T-Echo cannot power on after sleeping while on battery.  That was captured in detail in bug #1236 but that was marked of a duplicate of this one.  This problem still exists in v1.3.24.dff6915.  After the device is sleeping, pressing the bottom button results in fast flashing of a red LED (left one) and it does not wake.  Then, pressing the top button to reset works in the sense that the device resets and boots up successfully while on battery.  But isn't the bottom button alone supposed to turn it on? (not entirely sure what the real world difference is between reset and "turn on" when it is sleeping, or if there is any difference between "sleeping" and "off")