meshtastic/firmware #991: T-Echo - Freeze on BLE disconnect

sachaw, 2021-12-12:
From: #846
Freezing when Bluetooth disconnects (works "better" when connected to USB power but not with `is_always_powered` true

andrekir, 2021-12-12:
steps to reproduce:

- connect Android App to T-Echo normally;
- disable bluetooth on Android;
- wait.

visual cue --> T-Echo flashes red LED twice; no longer responds, no LEDs or bluetooth activity.

- before fw 1.2.46a: the device would freeze immediately after disconnecting bluetooth;
- after fw 1.2.46a: takes a couple of minutes.

sachaw, 2021-12-12:
Any chance you could have your device hooked up to a serial monitor and reproduce this?

andrekir, 2021-12-13:
finally caught one. BLE disconnect crash behaves different on USB power: instead of a freeze, the device restarts losing all data (nodes, gps loc, etc).

```
18:56:04 1684 BLE Disconnected, reason = 0x13
18:56:15 1695 [Power] Battery: usbPower=1, isCharging=1, batMv=4757, batPct=100
18:56:15 1695 [RadioIf] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=61, time 4628 ms
18:56:16 1696 [RadioIf] Lora RX (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0xb1 encrypted rxSNR=5.25)
18:56:16 1696 [RadioIf] AirTime - Packet received : 4628ms
18:56:16 1696 [Router] Add packet record (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0xb1 encrypted rxSNR=5.25)
18:56:16 1696 [Router] Using channel 0 (hash 0xb1)
18:56:16 1696 [Router] Expanding short PSK #1
18:56:16 1696 [Router] Installing AES128 key!
18:56:16 1696 [Router] decoded message (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:16 1696 [Router] handleReceived(REMOTE) (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:16 1696 [Router] Plugin nodeinfo wantsPacket=1
18:56:16 1696 [Router] Received nodeinfo from=0x3375d3c1, id=0x3c5f92c2, portnum=4, payloadlen=41
18:56:16 1696 [Router] old user !3375d3c1/p1337-Gateway/p13
18:56:17 1696 [Router] updating changed=0 user !3375d3c1/p1337-Gateway/p13
18:56:17 1696 [Router] Plugin nodeinfo considered
18:56:17 1696 [Router] Plugin routing wantsPacket=1
18:56:17 1697 [Router] Received routing from=0x3375d3c1, id=0x3c5f92c2, portnum=4, payloadlen=41
18:56:17 1697 [Router] Routing sniffing (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:17 1697 [Router] FIXME not implemented addRoute
18:56:17 1697 [Router] Rebroadcasting received floodmsg to neighbors (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:17 1697 [Router] Expanding short PSK #1
18:56:17 1697 [Router] Installing AES128 key!
18:56:17 1697 [Router] enqueuing for send (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim1 Ch0xb1 encrypted rxtime=1639421776 rxSNR=5.25)
18:56:17 1697 [Router] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=61, time 4628 ms
18:56:17 1697 [Router] txGood=43,rxGood=40,rxBad=0
18:56:17 1697 [Router] AirTime - Packet transmitted : 4628ms
18:56:17 1697 [Router] FIXME-update-db Sniffing packet
18:56:17 1697 [Router] Delivering rx packet (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:17 1697 [Router] Forwarding to phone (id=0x3c5f92c2 Fr0xc1 To0xff, WantAck0, HopLim2 Ch0x0 Portnum=4 rxtime=1639421776 rxSNR=5.25)
18:56:17 1697 [Router] Update DB node 0x3375d3c1, rx_time=1639421776
18:56:17 1697 [Router] Plugin routing considered
18:56:17 1697 [Screen] Screen: Joined: p1337-Gateway
18:56:18 1698 Lost phone connection
18:56:18 1698 (Client not yet interested in packets)Exception in thread rx:

--- exit ---
Traceback (most recent call last):
  File "/Users/andrekir/.platformio/penv/lib/python3.8/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/andrekir/.platformio/penv/lib/python3.8/site-packages/serial/tools/miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/Users/andrekir/.platformio/penv/lib/python3.8/site-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured
```

garthvh, 2022-05-05:
Is this still an issue?

kkwestt, 2024-05-24:
Not fixed 
BLE Disconnected, reason = 0x13
i use latest firmware 2.3.6 + ios latest

Saltubar, 2024-09-05:
Freezing still occurring on 2 different recently purchased T-Echo's using [Meshtastic Firmware 2.4.2.5b45303].
Symptoms are very similar to post earlier. I have seen this now twice on different T-echos both were on battery. No light are flashing, bottom button will not change screen or reboot on hold. No Bluetooth connection. Interestingly if I plug the device into the USB port on a windows machine the device comes back to life and operates normally and I can then disconnect the cable again and all is good. If I connect a USB to it with only a charger then no response.  Hope this helps