meshtastic/firmware #992: T-Echo - GPS sign gets flipped

sachaw, 2021-12-12:
From: #846
GPS sometimes flips the signs (+/-) on the coordinates i.e. (-16.696168, -50.359996 → +16.696168, +50.359996).

@andrekir any idea what conditions/timeframe causes this?

andrekir, 2021-12-12:
couldn't notice any triggers. the first sat lock is correct, but given enough time it always flips north/south or east/west coordinates.

runestone74, 2021-12-18:
The altitude also gets messed up sometimes
[image]
.

feh123, 2021-12-18:
I see the same with almost exactly the same altitude value and with the T-Echo showing 15 satellites. I also see cases where the longitude is correct but the latitude wrong by a factor of around 10. All these errors are sporadic - sometimes the gps fix is fine.
Is this due to an out of date bootloader program? Thanks.

garthvh, 2021-12-30:
I have long values that are missing the - and lat values that are periodically off by huge amounts.

runestone74, 2022-01-27:
Seems like the source code for Soft-RF GPS has a few more lines ;)
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/driver/GNSS.cpp
https://github.com/meshtastic/Meshtastic-device/blob/master/src/gps/GPS.cpp

I get almost instant GPS lock with a soft reset T-Echo with Soft-RF firmware, just running Meshtastic, where on Meshtastic it can take from 5 min until forever. I thought the internal antenna was bad, but it seems like Soft-RF GPS code is more robust.

Vertikar, 2022-02-02:
It looks like sometimes it just affects the GPS status screen, as the node distances are sometimes right on the screen, but look ok in the app. I have seen the node distances be reported wrong also though.

I took some screenshots and photos of it on the 1st.
Here's the photos of the t-echo
[image]
[image]
And a screenshot from the app:
[image]

runestone74, 2022-02-02:
I have now tried with latest .53 fw, no gps fix. Original Soft-RF fw, very quick GPS fix. Original LilyGo Meshtastic .42 fw (with no source) quick GPS fix, and correct time (0 unix time, 1970-01-01 01:00 in the iOS app though). So there is something not right with the newer fw and GPS readout for sure. It is not a hardware error at least.

runestone74, 2022-02-02:
UF2 Bootloader 0.5.0-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.9.0-22-g7cdeed54) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: LilyGo T-Echo
Board-ID: nRF52840-TEcho-v1
SoftDevice: S140 version 6.1.1
Date: May 13 2021

andrekir, 2022-02-02:
> It looks like sometimes it just affects the GPS status screen, as the node distances are sometimes right on the screen, but look ok in the app.

@Vertikar a bug is affecting MyNodeInfo updates on the android app. you should see correct/matching position data on all other nodes (but not on the top node card, ie. the node you are connected to). turning bluetooth off then on again will force an update.

@runestone74 a tip from Linar (from SoftRF) pointed out updating the bootloader is trivial (.uf2 drag-and-drop) in case you are interested: https://github.com/lyusupov/SoftRF/tree/master/software/firmware/binaries/nRF52840/Adafruit_nRF52_Bootloader

spoiler alert: makes no difference.

garthvh, 2022-02-02:
> I have now tried with latest .53 fw, no gps fix. Original Soft-RF fw, very quick GPS fix. Original LilyGo Meshtastic .42 fw (with no source) quick GPS fix, and correct time (0 unix time, 1970-01-01 01:00 in the iOS app though). So there is something not right with the newer fw and GPS readout for sure. It is not a hardware error at least. Also tried with Soft-RF fw, quick fix (within a minute) and directly switching to .53 fw, no fix, and wrong time.

I think we need to get the bootloader updated to a somewhat recent version before drawing any conclusions.  Based on what SoftRF did to support the TECHO there is a lot of code to write.

andrekir, 2022-04-18:
fixed with #1358

closed 2022-04-18 by andrekir

Vertikar, 2022-04-21:
Will this get pushed into a future 1.2 release, or only 1.3/2.0?