meshtastic/firmware #994: Feature Proposal/discussion - option to publish decrypted protobufs to MQTT

joshpirihi, 2021-12-12:
For my gateway node, I currently modify the code to publish decrypted protobufs to MQTT, then have a python script on my RPi that translates to other formats (eg raw number MQTT publish for battery %, an HTTP request for Traccar).

Given that some people are interested in doing this with their own MQTT server, how about we have an option to tell the device to decrypt the protobufs itself before publishing - it means we don't have to worry about decrypting it somewhere else first.  Obviously only really wanted if you are keeping it all in a lan or something.

Are there other MQTT solutions out there?  Or shall I open a pull request to add a "mqtt_publish_decrypted" preference?

andrekir, 2021-12-14:
MQTT feature is very new (especially the [Meshtastic-backend](https://github.com/meshtastic/meshtastic-backend) part) and contributions are very welcome.

I suggest checking if [disabling crypto](https://meshtastic.org/docs/developers/device/radio-settings#cryptography) like: `meshtastic --setchan psk 0` does what you want first.

also any ideas on what steps are missing to get an encrypted private MQTT gateway working?

joshpirihi, 2021-12-14:
I'm thinking of working out how to get the meshtastic device to publish plain data to MQTT, perhaps through a plugin if I can make it work.  That would mean we don't need a translator script to decode the protobufs which I think would be easier for users to set up.  I'll have a play with the code and see what I can make it do.

joshpirihi, 2021-12-14:
As for your question @andrekir, I looked at it for about 10 minutes and decided it was quicker to let the device do the decryption for me.  In theory its just a case of giving the python script the AES keys, and letting it decrypt with those.  But I have very little experience with crypto stuff

garthvh, 2022-01-26:
Changed to enhancement and assigned to @joshpirihi 

*[ ] Add a user preference we will default to false for MQTT Encryption
*[ ] Integrate Code change via bool to pull decoded or encrypted data

joshpirihi, 2022-01-26:
@garthvh thanks for the task list.  One question - do we really want to disable the mqtt encryption by default?  That will change device behaviour.  In the long run I reckon it's better to disable it (at least for custom mqtt servers).

I'd say that if a user has a custom mqtt server and is dealing with the encryption, they can probably handle this small change.  it shouldn't affect mesh bridging over mqtt.

I'm for disabling by default for a custom mqtt server.

garthvh, 2022-01-26:
@joshpirihi totally open to discussion but I think the default should be easy to set up and match the expectations for most MQTT clients which seems to be that the data is unencrypted.

I think you are our most active user of this feature.

caveman99, 2022-04-18:
seems implemented (and works well)- close?

garthvh, 2022-06-06:
Sure