meshtastic/firmware #997: T-echo node ID mismatch

jazap, 2021-12-15:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
The node ID on the unit's display does not match the node ID it reports to the Meshtastic App

**To Reproduce**
Steps to reproduce the behavior:
1. Press the unit's page flip button until you get tot he display showing the unit node ID on the upper right side of the display.
2. Open Meshtastic 1.2.45 and go to the node connection page.
3. Choose the node that shows available and connect.
4. Note the node ID you chose does not match the node ID on the unit, but you are connected to the unit.

mc-hamster, 2021-12-22:
@jazap Do you have screen shots or pictures of the problem? I'm trying to track this down.

mc-hamster, 2021-12-22:
Bug is likely in:

main-nrf52.cpp
   void getMacAddr(uint8_t *dmac)

jazap, 2021-12-23:
Jason Piazza
*operor vos ingredior ingredior?*

garthvh, 2021-12-28:
Can you check if this pull fixed your issue https://github.com/meshtastic/Meshtastic-device/pull/1017

andrekir, 2021-12-29:
installed latest firmware from artifact: https://github.com/meshtastic/Meshtastic-device/actions/runs/1631731764

no change. user id: **!495a8cdd**

[image]

andrekir, 2021-12-29:
correction, looks like the bug is fixed

didn't notice any changes immediately after factory_reset, but after a while owner/longname, user ID and Bluetooth ID (Meshtastic_89b0) all changed to reflect the same 4 letter ID (89b0) on the screen

[image] [image]

thebentern, 2021-12-29:
Woohoo! I was worried this one would be more insidious, but I like the fixes where I can just eliminate code

closed 2021-12-29 by sachaw

runestone74, 2021-12-30:
Is this related? The latest FW firmware-t-echo-1.2.50.41dcfdd.uf2 changed my device name from Meshtastic_1cf2 to Meshtastic_c773

garthvh, 2021-12-30:
Probably, check if that value properly matches the last four of your Mac and user.id now.  Sometimes takes a factory_reset to get synced back up.

runestone74, 2021-12-30:
meshtastic --set factory_reset 1 did not work. MAC: 73:c7:bf:9b:1c:f2 and still Meshtastic_c773 as the name, which seems derived from 73:c7 in the beginning of the MAC address

jazap, 2021-12-30:
That fixed it.
The node ID now matches the display in the Meshtastic app.

Jason Piazza
*operor vos ingredior ingredior?*