meshtastic/firmware #1001: add ability to monitor device on second serial port

mkinney, 2021-12-17:
**Describe the bug**
Add the ability to monitor the hardware via a second serial port, so we can capture information when there is an issue (like a reboot/lock up)

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to hardware using USB connection.
2. Connect to hardware using a different USB connection.
3. Run meshtastic cli thru first port.
4. Run/Show SerialConsole on second port.

**Expected behavior**
All debug output should go to 2nd serial port.

By default the Heltec nor T-Echo do not have a 2nd physical port, but they can be added by wiring up something like https://smile.amazon.com/gp/product/B01N47LXRA using https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/ as a guide

**Device info:**
 - Device model: HELTEC21 and TBEAM (not sure if TECHO has capablity)
 - Software Version: N/A

**Additional context**
Having this capability might help to troubleshoot intermittent issues and find the source of the problems.

mkinney, 2021-12-17:
After thinking about this for a bit, if we had this... we could do a soak-test where we run the same tests over and over until we see failure(s).

garthvh, 2022-10-13:
This already works with an external serial module, I can monitor on one and use the CLI on another

closed 2022-10-13 by garthvh