meshtastic/firmware #1003: Create an Auto channel selection

ianmagnone, 2021-12-17:
As the project moves from a more technical user base to more general users it would be helpful to have an Auto type channel selection that would allow the Meshtastic network to choose the balance of range and data rate.

Ideally it would start up in a long range, low data date mode to make sure it can establish a connection to the network and then if the devices can determine they are close enough they can increase the data rate at the cost of range. The goal would be to make the mesh as reliable as possible while being as quick as is reasonable.

RicInNewMexico, 2022-01-24:
I understand your reasoning as far as wanting to make it as reliable as possible, but from a spectrum usage perspective it would make more sense to go the opposite direction (attempt highest data rate / shortest transmission first and step down to progressively slower data rates as needed until communications is established). 

Doing auto-configuration of spread factor / data rate in this manner would also help ensure, for users operating in regulatory environments that have a duty cycle limitation, that the node isn't "wasting" transmission time when the traffic could have been passed faster (therefore allowing for more message traffic within the duty cycle). 

This is one of those issues that I don't think has any single "right answer". I agree that -some- automation of settings is a good thing, but with the massively varied levels of user skill level / involvement I think leaving the initial configuration as LongSlow and expecting manual tweaking to higher rates could end up leaving a lot of devices with unnecessarily long airtime usage.

ianmagnone, 2022-01-24:
Starting at highest data rate / shortest transmission and dropping down lower if need would work fine as well. Would that require the store and forward plugin though so that if the nodes initially connect on short/fast and then move apart breaking the link, it would be able to send the traffic once the connection is reestablished?

garthvh, 2022-08-06:
Closing, can always reopen if someone wants to take this on

closed 2022-08-06 by garthvh