meshtastic/firmware #1006: Ham Mode not working

mc-hamster, 2021-12-18:
OE1KBC — Yesterday at 4:33 PM
Hi, 1.2.49 ham-mode still working? ... i can not send text messages between two new TBEAM with 1.2.49 
back to 1.2.47 all ok

garthvh, 2022-01-23:
I think we also need a way to set short name for ham mode as any node with a license as the node name can't be differentiated from the others on the map.

garthvh, 2022-01-23:
I have also seen a couple of cases where the channel setting were only updated on devices under the control of the ham, and turning of encryption would require sharing those channel settings with the other users of the mesh.

sachaw, 2022-09-11:
Needs testing, also explicit instructions of what is required for ham mode to work. (Likely entire docs page)

caveman99, 2022-09-24:
as far as 1.2 is concerned its EOL. Is this a discussion point for 1.3/2.0?

garthvh, 2022-10-13:
This is all different now

garthvh, 2022-10-21:
This remains untested, in 1.3 setting ham mode should set lora.islicensed to true, set encryption to off on all channels and set long name to the ham callsign

thebentern, 2022-10-21:
Tested and it seems to work on 1.3.

1. Set LongName as call sign (and shortname based on that)
2. Dropped encryption from the primary channel
3. Set lora.isLicensed to true

closed 2022-10-21 by thebentern