meshtastic/firmware #1013: Sometimes when newly flashed nrf52 devices have different BLE Name Last 4 and nodeInfo.user.id last 4 do not match

garthvh, 2021-12-19:
The fields generated from the Mac address sometimes do not match on nrf52 devices

mc-hamster, 2021-12-19:
Could you provide screen shots or pictures of the two locations?

garthvh, 2021-12-19:
I have feedback from some users and I have seen it a couple of times, I just asked them for screenshots of meshtastic --info

mc-hamster, 2021-12-19:
Source of truth for Mac address should be:

main-nrf52.cpp
  void getMacAddr(uint8_t *dmac)

Anything that doesn't use that method (or its return) may cause the problem we're looking for.

garthvh, 2021-12-20:
Good news is one of the affected users is on @thebentern local network

mc-hamster, 2021-12-22:
This post suggests that our use of sd_ble_gap_addr_get and DEVICEADDR should be reviewed:

https://devzone.nordicsemi.com/f/nordic-q-a/50460/what-is-the-relationship-between-the-ble-address-and-the-various-ficr-registers

thebentern, 2021-12-22:
I read some similar posts on the Nordic community. It would seem like that method doesn't return the same output for before vs after ble broadcast is initiated. Unless I'm mistaken, I wonder if we can get away with just removing the if statement and simply using the code in the else statement to pull the "mac" off of
`NRF_FICR->DEVICEADDR`.  These apparently are just essentially random ids assigned at time of production.

garthvh, 2021-12-29:
Verified fixed on a 5005 and 19003 by me and an affected T-Echo by @andrekir

closed 2021-12-29 by garthvh