meshtastic/firmware #1014: Changing channel settings require device reset

jaykayenn, 2021-12-21:
Changing channels causes radio SNR to drop drastically, until the devices are reset. 

Steps:
- When devices are first switched on, all is working normally. Node list shows 10+db and signals are clearly seen in the air via an SDR monitor. 
- Change the channel on all devices (QR code)
- Node list now shows all devices at negative db values. Signals no longer seen on SDR monitor. Messages are only successfully transmitted if devices are right next to each other.  Sometimes not even then. 
- Press reset button and restart devices. 
- Devices now work correctly. Node list shows 10+db again, and signals seen in SDR. Normal range performance restored. 

**Expected behavior**
Should be able to set/change channels without having to reset the device. Users may be frustrated by drastic drop in performance after activating QR code. 
If reset is required and unavoidable, make it very clear to the user, or force the device to self-reset after channel change. 

**Device info:**
 - Device model: Tbeams and TLORA 2.1
 - Firmware 1.2.49

**Smartphone information (if relevant):**
- Poco F1, LineageOS, Android 10, app 1.2.51
- Samsung Tab A7, stock Android 10, app 1.2.51

jaykayenn, 2021-12-30:
Still persistent on firmware 1.2.50

jaykayenn, 2022-01-02:
Latest test results, firmware 1.2.50

- Setup 4 nodes (2 TBeam, 2 TLora 2.1)
- All nodes switched on with default LongSlow. All nodes see each other with 10+db
- Switch nodes to MedSlow one by one. 
- As each node is set to MedSlow, that node now shows as 0db or less. 
- All nodes are eventually set to MedSlow. All nodes appear with 0db or less. 
- Reset nodes one by one. 
- As each node is reset, that node then appears as 10+db again. 
- All nodes are eventually reset, all showing 10+db again. 

Channel settings by Android app 1.2.52
Nodeinfo dbi observed via app and CLI.

geeksville, 2022-01-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/stock-tbeam-antenna-15db-snr/4223/6

garthvh, 2022-01-24:
@thebentern can we run a device reset after channel changes are made to ensure there is a consistant experience?

ernax78, 2022-02-02:
I had approximately the same experience with python:

1) I set a new speed --ch-longfast
2) I copied the URL to the other devices
3) No dice (devices would not see eachother)

After deeper inspection, I saw the bitrate had remained the same after changing the speed (--ch-longfast) and --info was still  "bitrate": 17.08847.

When rebooting the device, "bitrate" was increased to the one reflecting --ch-longfast AND the URL was different.

After implementing this newly generated URL around my devices and rebooting, the mesh started to work (except on my Eltec but that is probably another story).

To reproduce:
1) Get URL using --info
2) Set a new speed on primary channel
3) Get URL using --info
4) Reboot device
4) Compare URL to old URL before reboot.

This was tested a while ago on firmware 1.2.50 but I dont think it has changed.

geeksville, 2022-05-04:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/node-to-node-bluetooth-and-spreading-factor/5784/8

mc-hamster, 2022-05-05:
Good call out.

This should trigger the device to restart.

garthvh, 2022-06-14:
I still get reports about this, are we rebooting the device when channel settings change yet?

ernax78, 2022-06-16:
I dont know if this was already fixed, but if we change the channel (thus generating a  new qr code), why not (optionally, default: yes) wipe the "nodes" database at the same time? The nodes you found on the previous encryption / channel will not be able to communicate with the node anyways due to other physical and cryptographical mechanism. It will also provide better feedback when re-establishing the mesh on the new parameters. (Maybe this is already implemented in some sort, I tried to search)

mc-hamster, 2022-06-22:
Device code purposely doesn't reset the device. Doing so would not be proper behavior.

See the comment on NodeDB.cpp for radioGeneration.

Fix is to properly configure the lora radio when the modem configuration is changed.

mc-hamster, 2022-06-22:
Added some debug output to monitor the power configuration when the radio is configured / reconfigured.

    DEBUG_MSG("Frequency set to %f\n", getFreq());    
    DEBUG_MSG("Bandwidth set to %f\n", bw);    
    DEBUG_MSG("Power output set to %d\n", power);

geeksville, 2022-07-19:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/meshtastic-1-3-25-july-11-public-preview/6124/10

mc-hamster, 2022-07-19:
> Can we reboot automatically from the device after changing lora or channel config? With a default of UNSET I no longer see any cases where a reset is not required for messaging to work.

We shouldn't reboot. There's a lot of code that tracks configuration changes and is supposed to reload data structures.

If we do have to reboot, that's a bug.

We can reboot in the short term as a work around, but that doesn't hide that there is a defect in there.

garthvh, 2022-08-06:
We need to test this on 1.3, changing region no longer requires a reboot for me most of the time.

sachaw, 2022-08-07:
Changing channels settings fork for me without reboots

sachaw, 2022-09-11:
Auto reboot now implemented.

closed 2022-09-11 by sachaw