meshtastic/firmware #1019: Send Low Battery Lora Message for Devices with Simple Chargers

garthvh, 2021-12-23:
Several devices supported by Meshtastic do not have sophisticated charging circuitry and without protected cells could deep discharge batteries.  

- Heltec (ESP32)
- LORA32 (ESP32)
- RAK 5005 and 19003 (nrf52)

Proposal would be to send a message from a device somewhat before the battery reaches a potentially damaging level for these devices.  Time thresholds would probably need to be different based on the chip used as 25% with a nrf52 could last much longer than an ESP32 device.

mc-hamster, 2021-12-23:
Battery level (battery_level) is already included in the position packet. Can that be used?

garthvh, 2021-12-23:
> Battery level (battery_level) is already included in the position packet. Can that be used?

Seems like the right value to use, this was described to me as how large agricultural lora networks manage Lipo powered remote nodes.

jaykayenn, 2021-12-24:
battery_level for TLora seems to be scaled incorrectly. My batteries never reach beyond 60%+, and USB wall power shows as 75%. It does not trigger POWERED mode.

mc-hamster, 2021-12-25:
Closing this issue. Please re-open if the data is not being sent.

closed 2021-12-25 by mc-hamster