meshtastic/firmware #1023: Update esp8266-oled-ssd1306

mc-hamster, 2021-12-26:
Our use of esp8266-oled-ssd1306 is old

thebentern, 2021-12-27:
@mc-hamster, I almost wonder if we should start spot checking our lib versions before every release or maybe on a monthly schedule? Seems like we've run into a number of these recently.

mc-hamster, 2021-12-27:
Reopening. The Linux build doesn’t run with this change

mc-hamster, 2021-12-27:
> @mc-hamster, I almost wonder if we should start spot checking our lib versions before every release or maybe on a monthly schedule? Seems like we've run into a number of these recently.

I was actually thinking about adding the snapshot date in the platform.ini file so we can be easily audited.

sachaw, 2021-12-27:
Wondering if we could pipe the output of one of these commands: https://docs.platformio.org/en/latest/core/userguide/lib/cmd_update.html
as a warning in the action if there are any outdated libraries, then it will show in the output once it's run.
An example: https://github.com/meshtastic/meshtastic-web/actions/runs/1618137015

thebentern, 2021-12-27:
That's a good step in the right direction. 

I was hoping that pio would have some lib check api or cli command to let us know if there are newer versions of a package available like most package managers do, but I couldn't find anything there. If we want something like that, we'd have to cobble up our own script to do so.

caveman99, 2022-01-27:
fixed with https://github.com/meshtastic/Meshtastic-device/pull/1139 - of course the scanning for outdated libs is still an issue

closed 2022-04-26 by thebentern