meshtastic/firmware #1025: Full battery test timeline for my Heltec board

newtocrypto, 2021-12-26:
Full battery test timeline:

2145 25DEC - Orange light went out after a few hours indicating a fully charged battery. With the USB still connected the battery read 40% charged 3.65v on the screen, I did not test from the back of the terminal.

2150 - USB disconnected the battery read 38% charged 3.63v on the screen, and from the back of the terminal it read 4.13v 

2151 - Battery disconnected separated from the device it read 4.13v

2152 - Plugged battery back in and left it alone

0945 26 DEC - I sent the device message #1 to wake it up, message received. On battery power only it read 5% charged 3.32v on the screen, and from the back of the terminal it read 3.82v

1045 - I sent the device message #2 to wake it up, message received. On battery power only it read 2% charged 3.28v on the screen, and from the back of the terminal it read 3.80v

1145 - I sent the device message #3 to wake it up, message received. On battery power only it read 0% charged 3.25v on the screen, and from the back of the terminal it reads 3.70v

1250 - I sent the device message #4 to wake it up, message NOT received.
I tapped the top button and the device woke up, the uptime timer had started over, I sent message #5 and it was received. 
On battery power only it read 3% charged 3.25v on the screen and then died after a minute or so.
Maybe this part I did wrong but I was only able to get 0.91v from the back of the terminal.
I disconnected the battery and it read 3.79v

Using a 1000 mAh battery and stock settings for 1.2.45 beta firmware on a Heltec v2.1 board

My discord is @anchormetals and I was asked to put this here by one of the community members, hopefully, I did it correctly.

jaykayenn, 2021-12-26:
More info on forum thread: https://meshtastic.discourse.group/t/heltec-esp32-only-charging-to-40-at-3-65v/4515

newtocrypto, 2021-12-27:
OK, so I looked at the link @jaykayenn sent...

I used the 3.70 firmware from the link, my math went like this…
Charged battery from the terminal read 4.15v and one the screen 3.63v
3.63 / 3.2 = 1.134375
4.15 / 1.134375 = 3.658
I rounded my 3.658 up to 3.7 and if I used 4.2 for the terminal it would have come out to 3.702

I updated the new firmware, orange light comes on, and the battery starts to charge.
With the USB still connected the battery cycles between 0% + U at 4.23v and 100% at 4.20v on the screen, I did not test the back of the terminal. The battery indicator also cycles between a weird double line symbol and full.

When the orange light went out, indicating charging complete, and with the USB still connected the battery cycled between 0% + U at 4.23v and 100% at 4.20v on the screen, I did not test the back of the terminal. The battery indicator also cycles between a weird double line symbol and full.

With the USB disconnected and on battery power only the screen read 99% at 4.19v and 4.15v from the back of the terminal, and the battery indicator is full.

I will leave it unplugged and will do the same test as before...

jaykayenn, 2022-01-02:
This applies to TLora32 v2.1-1.6 as well, same vbat discrepancy between firmware and actual battery.

tropho23, 2022-01-10:
I have two Heltec LoRa32 v2.1 boards and have the same issue.  I'm using 3,000mAh LiPo MakerFocus batteries but the issue remains and I don't ever see more than 52% charge noted on the screen for one Heltec, and no more than 53% on the other.  This is after the red battery charge LED on the board has gone out, indicating the battery is fully charged.

I have not done a runtime test but presume runtime would be 2.5-3x newtocrypto's results using a 1,000mAh battery.

mchisari, 2022-01-16:
I confirm, when the orange light turns off the display shows 4.0V. Maybe some adjusting to voltage formula could fix?

garthvh, 2022-01-24:
Closing will track extensive Heltec battery research / issues on https://github.com/meshtastic/Meshtastic-device/issues/769

closed 2022-01-24 by garthvh

geeksville, 2022-02-21:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/how-to-pair-headless-ttgo-lora32-no-display/3296/15

newtocrypto, 2022-02-22:
I upgraded my Heltec v2.1 boards to firmware-1.2.54.288f2be and the battery issue seems to be fixed with this update...

The battery indicator reads full with a 4.2v reading on the battery and on the screen.