meshtastic/firmware #1032: Node hangs when the automatic screen carousel (#937) is enabled

osmanovv, 2021-12-28:
The device hangs when the automatic screen carousel (#937) is enabled with a small number of seconds. E.g.
```
meshtastic --set auto_screen_carousel_secs 7
```

It may work for several hours and suddenly freeze.

There may be several reasons:
* Automatic screen carousel behavior dramatically increases the number of packets transmitted over the radio network - it sends a new ping packet to the displayed node.
https://github.com/meshtastic/Meshtastic-device/blob/33769b8657eee58282bc3b53eefe36945dad3520/src/graphics/Screen.cpp#L623-L626
* Peak load associated with increasing the frequency of the MCU when scrolling the screen (#983).
* Other reason, as I have noticed sudden hangs on button press on previous firmware versions.
* Two and more nodes with the automatic screen carousel can lead to a network flood as they frequently ping the nodes.

I reproduce it on TBEAM and DIY nodes with 1.2.49.

@thebentern, @mc-hamster, please take a look.

thebentern, 2021-12-28:
Good catch! There's certainly a number of moving pieces here.
@mc-hamster, for the first item and I'm sure you'll agree, I would propose that we move any mesh logic out of the screen class... network layer code polluting the presentation layer is no bueno. 

Do you have any thoughts about how to handle something like this on an interval elsewhere? The only other place we call that method is in powerFSM. Two minutes also seems like a pretty arbitrary interval that may not suit all networks / use cases.

mc-hamster, 2021-12-28:
if (sinceLastSeen(n) > 120) {
            service.sendNetworkPing(displayedNodeNum, true);
        }

Shouldn't be in drawNodeInfo at all. That could be taken out. There's no intuitive user trigger that would indicate the behavior and no feedback that the information is pending an update. This very well could flood a network if someone pages through the screen on a device.

I could see this being triggered on a client (iPhone, iOS, web) but not from the device.

To answer the question about where periodic activities can be placed ... one is to inherit from concurrency::OSThread (see airtime for an example) if no network access is needed or to create a plugin if accessing the mesh is required.

mc-hamster, 2021-12-28:
> Peak load associated with increasing the frequency of the MCU when scrolling the screen 

This wouldn't increase peak load, work on the firmware are timer based and the timer doesn't change with cpu speed.

mc-hamster, 2021-12-28:
> * Two and more nodes with the automatic screen carousel can lead to a network flood as they frequently ping the nodes.

This may be the cause of reports we've had of nodes missing messages when someone is running a test of multiple devices. The network is just flooded.

Maybe this suggests we should have some sort of 'network saturation' metric. That won't be hard to implement.

0% - unsaturated
>75% (or whatever value) - saturated

thebentern, 2021-12-28:
Good idea. That could be a useful diagnostic for knowing if something like this is occurring elsewhere.

osmanovv, 2022-01-10:
Nodes are still hanging with the screen carousel enabled on 1.2.50.

garthvh, 2022-01-24:
Not happening anymore on 1.2.52 with 15 second screen updates.

closed 2022-01-24 by garthvh