meshtastic/firmware #1034: network saturation metric

mc-hamster, 2021-12-28:
For background, see: https://github.com/meshtastic/Meshtastic-device/issues/1032

Create a network saturation metric. This counts sent and received packets over a short time window.

Within the window, use something like a boxcar or moving average to create the value.

Data can be gathered by utilizing existing airtime methods.

Saturation should be displayed on device screen and added to node info.

Propose that this be a 90 second measurement window. On long/slow, worse case packet takes about 15 seconds.

linagee, 2021-12-28:
If possible, this should be any packet that is detected. (Basically anything matching the LoRa preamble. That way TTN/Helium/etc traffic would count.)

If RSSI from the radio chip shows too high of a value, it is also unusable airtime even if there's no preamble detected. (Noise would interfere with communication and indirectly cause "utilization".)

It would be quite interesting to push this up to the cloud if the "Anonymous Statistics" box is checked. (Along with rough GPS data, this could be used to make a heatmap of LoRa saturation for a given area / frequency / spreading factor.)

Maybe there is some standard way Semtech has in mind to do this already?

Battery vs non-battery should be considered. Is the code already optimized to only wake up the ESP32 if a message is seen for us? That would be battery friendly. If we want to relay all of this stuff to the phone, more power hungry. (Even power for BLE.)

mc-hamster, 2021-12-28:
> If possible, this should be any packet that is detected. (Basically anything matching the LoRa preamble. That way TTN/Helium/etc traffic would count.)

Got that already in the system a few months ago. :)

closed 2021-12-29 by mc-hamster