meshtastic/firmware #1035: Add Tap back like Feature to Messaging

garthvh, 2021-12-29:
Messaging clients for iOS and Android and many others all contain a tap back feature that allows users to quickly respond with preselected emoji options to a message.  

@mc-hamster and I were discussing this potential feature and it seemed worth capturing.

* Could potentially be implemented as a separate packet type with a replyTo messageId and a single unicode character in the payload.
* Could also be integrated into the text message protocol directly so If the "reply to" message id isn't set (eg: regular packets) then there's no cost over the wire. If it's set, then the payload could just be the unicode character for the tap back. We need to consider the impacts of any additions to the text message proto affecting maximum payload size
* An ASCII_BELL character could be utilized as the character used to send an "emergency button" message to all users on the main broadcast.

mc-hamster, 2021-12-29:
reply_id is the message id of the message you're replying to.
is_tapback determine what kind of reply this is. if false, it's a text reply. if true, it's a emoji reply.

if reply_id = something and is_tapback = false, then this is a reply like in ios or discord.
if reply_id = something and is_tapback = true, then the payload should only be a single character. Emoji or otherwise.
if reply_id = undefined and is_tapback = true, then the use wants to send a single large emoji or in the case of our discussion of ASCII_BELL, then the user wants to send an alert.

Change to protobuf to support this:
  https://github.com/mc-hamster/Meshtastic-protobufs/commit/67232d9c95f24cfbd5c6699701fb726eaecce216

mc-hamster, 2021-12-29:
Closing. Device support for this will be included in this week's release.

Clients will also need to adopt the feature.

mc-hamster, 2021-12-29:
This will be backward compatible for clients that don't explicitly support this as it'll just show up as normal messages.

I'll hold off on supporting this in S&F for the time being.

closed 2021-12-29 by mc-hamster