meshtastic/firmware #1039: TLora32 2.1 does not send batteryLevel to mesh

jaykayenn, 2021-12-29:
Remote nodes running on TLora32 2.1 always appear in nodelist with no batteryLevel. 

- Setup a network of Meshtastic nodes connected to a channel
- View list of nodes via CLI or phone app
- See that any remote nodes running on TLora do not have batteryLevel in the Nodeinfo
- TLora nodes will show their own batteryLevel when viewing node list locally, but will not see batteryLevel of other TLora.
- TBeams do not have this issue. TLora will see remote TBeam batteryLevels. 

```
My info: { "myNodeNum": 633505936, "numBands": 10, "firmwareVersion": "1.2.49.5354c49", "rebootCount": 47, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }
Nodes in mesh:  {'num': 633505936, 'user': {'id': '!25c28890', 'longName': 'Mini1', 'shortName': 'Mn1', 'macaddr': 'THUlwoiQ', 'hwModel': 'TLORA_V2_1_1p6'}, 'position': {'batteryLevel': 63}, 'lastHeard': 1640766975}  {'num': 633505116, 'user': {'id': '!25c2855c', 'longName': 'Mini2', 'shortName': 'Mn2', 'macaddr': 'THUlwoVc', 'hwModel': 'TLORA_V2_1_1p6'}, 'position': {}, 'lastHeard': 1640766715, 'snr': 9.25}
```

mc-hamster, 2022-01-11:
Reference: https://resource.heltec.cn/download/WiFi_LoRa_32/

mc-hamster, 2022-01-11:
schematic:

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/T3_V1.6.1.pdf

mc-hamster, 2022-01-11:
the board says t3_v1.6.1, the purchase link says v2.1 1.6.1 and the schematic says t3 v1.6.1

board was purchased from our supported page

https://meshtastic.org/docs/hardware/supported/lora

mc-hamster, 2022-01-11:
Take your measurement at IO35
So, that's one voltage measurement at io35 and another at the battery
jkn — Today at 11:35 PM
4v -- 1.865v

thebentern, 2022-01-11:
The T3 V1.6 is the V2.1 1.6. I know, it's confusing right?

> TTGO LoRa32 V2.1.6 (aka T3 V1.6, aka LoRa32 V2.1 release 1.6):
Is labeled T3 V1.6, 20180506 on the PCB.
Similar to V2.1.5 with changes and improvements:
Changes in pins / GPIOs. Uses GPIO23 for LoRa RST. Uses GPIO25 for onboard LED. Uses GPIO35 as ADC for battery voltage measurement (uses 100k/100k divider).

This thread has a pretty comprehensive summary of the TLoRA / LoRA32 board variants.
https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-3/18436
I think there may be some more recent revisions available that aren't covered, especially given the fact that Lilygo has made small changes to the T-Beam such as swapping out usb serial ICs.

geeksville, 2022-01-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/ttgo-lora32-t3-v1-6-1-no-oled/4742/3

jaykayenn, 2022-01-11:
> This issue has been mentioned on **Meshtastic**. There might be relevant details there:
> 
> https://meshtastic.discourse.group/t/ttgo-lora32-t3-v1-6-1-no-oled/4742/3

That thread was made in error, and is now talking about something else unfortunately.

garthvh, 2022-02-04:
Tested that batteryLevel is sent in position packets to the mesh as expected 1.2.53.  May be an issue in the python api if the display is wrong.

closed 2022-02-04 by garthvh