meshtastic/firmware #1040: ExternalNotificationPlugin doesn't work

prampec, 2021-12-29:
**Describe the bug**
ExternalNotificationPlugin was set up correctly but tone was not played at message receive.

**To Reproduce**
Steps to reproduce the behavior:

1. Connected active piezo buzzer positive lied to 3v3 pin, negative lied to pin “13”.
2. Set up configuration with python CLI so it reflect the settings below.
3. Restarted the device.
5. Sent any text messages from other device.
6. Message arrived, but not tone is produced.

**Expected behavior**
Audible tone on message arrival.

**Used python CLI commands to configure the plugin**
meshtastic --set ext_notification_plugin_enabled true
meshtastic --set ext_notification_plugin_alert_message true
meshtastic --set ext_notification_plugin_output_ms 13
meshtastic --set ext_notification_plugin_output_ms 1500

**Settings reflected by python CLI**
ext_notification_plugin_enabled: True
ext_notification_plugin_alert_message: True
ext_notification_plugin_active: False
ext_notification_plugin_output_ms: 1500
ext_notification_plugin_output: 13

**Device info:**
 - Device model: TTGO Lora v1.0
 - Software Version 1.2.47

**Additional context**
Also have tried it with current GIT version.
Also tried with a simple Arduino sketch to pulling the pin LOW, and it the buzzer beeps. So the hardware setup is correct.

geeksville, 2021-12-29:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-plugin-externalnotificationplugin/2328/26

prampec, 2022-01-01:
Debug log:
```
??:??:?? 1019 [Router] Plugin ExternalNotificationPlugin wantsPacket=1
??:??:?? 1019 [Router] packet on wrong channel, but can't respond (id=0xbffda5ce Fr0x78 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=1 rxSNR=9.5)
```

prampec, 2022-01-01:
I'm not completely sure what channel-binding is about to mean.
In this case ExternalNotificationPlugin was bound to channel 'gpio', while packets are received to empty channel name.

So, plugins with specified channel-binding should exclusively receive messages to that very channel? In this case we should not set binding to ExternalNotificationPlugin.
Channel-binding is some kind of hint pre-filter messages? Than we might consider defining a set of channels, that are accepted. And empty channel name should be an item in this list for ExternalNotificationPlugin.

That's being said I have provided a naive workaround for the problem as a pull request:
https://github.com/meshtastic/Meshtastic-device/pull/1048

mc-hamster, 2022-01-02:
Thanks for the PR!

Looks like this hasn't been used in a while. Appreciate the effort!

PR #1048 (2022-01-02): Channel-binding filter accepts empty channel names.
This is a workaround that fixes #1040.

In this naive solution packets are accepted for plugins having a boundChannel, when the channel of the packet is actually empty (or doesn't have channel specified at all).

Please consider, that this might not be the goal of the channel-binding of a plugin. See #1040 for details!
closed 2022-01-02 by mc-hamster